dbuezas/esphome-cc1101 #4: not able to send

ageurtse, 2022-08-16:
hello i have aa cc1101 hooked up to a wemos D1 mini. 

when hooking up i could recieve some data. 

but when i resend this data there is nothing happening, the lights wont turn on.
what could be wrong, how to find out what is wrong. 

below is my yaml file

`# default template name, the first number in the name is the ending of the ip adres
# --- start default settings ---
esphome:
  name: 118-rf-tranceiver
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"
    
esp8266:
  board: d1_mini_lite

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "rf fallback"
    password: !secret esphome_pwd
  
  manual_ip:
    static_ip: 192.168.1.118
    gateway: 192.168.1.1
    subnet: 255.255.255.0

captive_portal:

# Enable logging
logger:

api:
  password: !secret esphome_api

ota:
  password: !secret esphome_ota

web_server:
  port: 80

# --- End default settings ---

# --- Start of custom sensors ---

sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        D5, // SCK
        D6, // MISO
        D7, // MOSI
        D8, // CSN
        D1, // GDO0
        D2, // GDO2
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: D1 # This is GDO0
    carrier_duty_percent: 100%

remote_receiver:
  - pin: D1 # This is GDO0
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump: rc_switch
      
binary_sensor:
  - platform: remote_receiver
    name: Garage
    rc_switch_raw:
      code: "0011000100001110101001111000"
      protocol: 1
button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: "0011000100001110101001111000"
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: "0011000100001110101001111000"
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();
      
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: "0011000100001110101001111000"
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: "001100010000111010100111100"
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();`

dbuezas, 2022-08-16:
Try raw and paste the logs here:
https://dbuezas.github.io/esphome-remote_receiver-oscilloscope/

It may be that it is recognized as "protocol 1"  but it isn't. The full raw mode is more verbose but it doesn't assume any particular timing.

It can also be that your garage uses rolling codes, in which case the codes change all the time and transmitting won't work (unlikely)

ageurtse, 2022-08-16:
I did read the raw data but at every key press the data isn't the same. 
Tommorow i will send the output of the raw data.

dbuezas, 2022-08-16:
Make a couple of readings and paste the full log there, you'll be able to visualize it and see if they are truly different each time or just slight timing deviations (whis is normal and expected).
If they do look the same in the plot, the n transmitting should work fine with those approximate flank timings.
Much luck!

ageurtse, 2022-08-17:
Past the raw messages in your tool, and they all look the same. 

```[06:37:56][D][remote.raw:028]: Received Raw: 364, -1120, 389, -1125, 1112, -392, 1116, -385, 385, -1128, 368, -1140, 365, -1126, 1120, -388, 359, -1152, 344, -1141, 381, -1123, 371, -1141, 1105, -397, 1124, -382, 1105, -398, 350, -1158, 1103, -397, 364, -1141, 1104, -398, 350, -1140, 
[06:37:56][D][remote.raw:041]:   374, -1143, 1115, -383, 1106, -396, 1124, -383, 1106, -397, 374, -1140, 351, -1145, 350, -1140, 375
[06:37:56][D][remote.raw:028]: Received Raw: 386, -1113, 392, -1121, 1110, -394, 1119, -389, 360, -1132, 368, -1148, 360, -1156, 1093, -385, 385, -1129, 369, -1124, 386, -1130, 369, -1125, 1111, -392, 1116, -388, 1112, -394, 368, -1141, 1111, -393, 369, -1144, 1090, -400, 374, -1143, 
[06:37:56][D][remote.raw:041]:   365, -1140, 1103, -399, 1102, -405, 1102, -400, 1101, -380, 375, -1145, 350, -1138, 375, -1144, 364
[06:37:56][D][remote.raw:028]: Received Raw: 395, -1116, 390, -1116, 1120, -381, 1114, -394, 369, -1144, 357, -1137, 369, -1146, 1116, -370, 369, -1144, 356, -1136, 370, -1143, 357, -1137, 1121, -388, 1115, -394, 1120, -389, 360, -1131, 1118, -387, 361, -1152, 1119, -386, 358, -1153, 
[06:37:56][D][remote.raw:041]:   342, -1164, 1084, -395, 1122, -384, 1109, -396, 1123, -383, 381, -1121, 370, -1132, 382, -1123, 370
[06:37:56][D][remote.raw:028]: Received Raw: 372, -1116, 390, -1140, 1097, -388, 1110, -394, 369, -1145, 364, -1140, 347, -1143, 1114, -405, 353, -1144, 364, -1140, 354, -1144, 365, -1141, 1107, -397, 1098, -407, 1106, -398, 349, -1164, 1105, -399, 351, -1140, 1105, -410, 357, -1134, 
[06:37:56][D][remote.raw:041]:   369, -1145, 1114, -376, 1127, -374, 1114, -406, 1115, -390, 350, -1138, 376, -1145, 350
[06:38:59][D][remote.raw:028]: Received Raw: 387, -1131, 368, -1149, 1114, -393, 1093, -388, 387, -1126, 367, -1149, 361, -1129, 1118, -387, 386, -1126, 366, -1149, 361, -1128, 369, -1146, 1110, -394, 1121, -389, 1111, -393, 369, -1145, 1108, -396, 345, -1143, 1114, -405, 352, -1147, 
[06:38:59][D][remote.raw:041]:   373, -1141, 1103, -400, 1098, -406, 1105, -398, 1099, -407, 353, -1146, 372, -1140, 352, -1148, 372
[06:38:59][D][remote.raw:028]: Received Raw: 396, -1118, 390, -1136, 1097, -402, 1106, -399, 373, -1142, 365, -1141, 351, -1146, 1123, -383, 353, -1147, 372, -1140, 353, -1146, 372, -1140, 1106, -397, 1124, -383, 1105, -397, 373, -1140, 1103, -398, 374, -1140, 1103, -398, 351, -1139, 
[06:38:59][D][remote.raw:041]:   375, -1144, 1115, -384, 1108, -396, 1122, -384, 1112, -392, 371, -1141, 379, -1123, 372, -1141, 353
[06:38:59][D][remote.raw:028]: Received Raw: 375, -1140, 373, -1145, 1101, -405, 1102, -399, 351, -1140, 374, -1147, 350, -1139, 1126, -374, 379, -1139, 373, -1142, 364, -1141, 352, -1145, 1123, -383, 1105, -396, 1121, -385, 380, -1123, 1121, -385, 381, -1120, 1115, -412, 359, -1122, 
[06:38:59][D][remote.raw:041]:   390, -1142, 1109, -394, 1095, -412, 1110, -395, 1094, -412, 358, -1147, 366, -1125, 387, -1124, 366
[06:38:59][D][remote.raw:028]: Received Raw: 387, -1127, 370, -1116, 1134, -370, 1120, -387, 383, -1124, 370, -1142, 381, -1121, 1121, -386, 382, -1122, 370, -1142, 380, -1120, 371, -1131, 1134, -369, 1140, -387, 1111, -394, 369, -1142, 1109, -369, 370, -1141, 1135, -369, 370, -1140, 
[06:38:59][D][remote.raw:041]:   381, -1121, 1139, -388, 1109, -395, 1119, -388, 1110, -394, 369, -1117, 383, -1123, 370```

the first 4 are from the first button press, and the last 4 are from the 2e button press. 

but upon sending the raw data back the light's won't go on. 

below the alterd yaml
```# default template name, the first number in the name is the ending of the ip adres
# --- start default settings ---
esphome:
  name: 118-rf-tranceiver
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"
    
esp8266:
  board: d1_mini_lite

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "rf fallback"
    password: !secret esphome_pwd
  
  manual_ip:
    static_ip: 192.168.1.118
    gateway: 192.168.1.1
    subnet: 255.255.255.0

captive_portal:

# Enable logging
logger:

api:
  password: !secret esphome_api

ota:
  password: !secret esphome_ota

web_server:
  port: 80

# --- End default settings ---

# --- Start of custom sensors ---

sensor:
[... 26 lines cut ...]
      - raw
      
binary_sensor:
  - platform: remote_receiver
    name: Garage
    rc_switch_raw:
      code: "11111100"
      protocol: 1
      
button:
  - platform: template
    name: Garage
    on_press:
      - remote_transmitter.transmit_raw:
          code: [364, -1120, 389, -1125, 1112, -392, 1116, -385, 385, -1128, 368, -1140, 365, -1126, 1120, -388, 359, -1152, 344, -1141, 381, -1123, 371, -1141, 1105, -397, 1124, -382, 1105, -398, 350, -1158, 1103, -397, 364, -1141, 1104, -398, 350, -1140, 374, -1143, 1115, -383, 1106, -396, 1124, -383, 1106, -397, 374, -1140, 351, -1145, 350, -1140, 375]
      - lambda: get_cc1101(transciver).endTransmission();
      - remote_transmitter.transmit_raw:
          code: [364, -1120, 389, -1125, 1112, -392, 1116, -385, 385, -1128, 368, -1140, 365, -1126, 1120, -388, 359, -1152, 344, -1141, 381, -1123, 371, -1141, 1105, -397, 1124, -382, 1105, -398, 350, -1158, 1103, -397, 364, -1141, 1104, -398, 350, -1140, 374, -1143, 1115, -383, 1106, -396, 1124, -383, 1106, -397, 374, -1140, 351, -1145, 350, -1140, 375]
[... cut ...]
      - lambda: get_cc1101(transciver).endTransmission();
      - remote_transmitter.transmit_raw:
          code: [387, -1127, 370, -1116, 1134, -370, 1120, -387, 383, -1124, 370, -1142, 381, -1121, 1121, -386, 382, -1122, 370, -1142, 380, -1120, 371, -1131, 1134, -369, 1140, -387, 1111, -394, 369, -1142, 1109, -369, 370, -1141, 1135, -369, 370, -1140, 381, -1121, 1139, -388, 1109, -395, 1119, -388, 1110, -394, 369, -1117, 383, -1123, 370]
      - lambda: get_cc1101(transciver).endTransmission();```

dbuezas, 2022-08-17:
Ok, the plot looks like this should work, but you are missing the startTransmission lambda calls now. That may just be it!
Also, make sure to pick the longer pattern from the ones you received. Good luck!

ageurtse, 2022-08-17:
stupid me, had to sleep longer :-)

after altering nothing happens, i think i have to wait for a second CC1101, so i could monitor what is sending. 
since there is still no communication.

dbuezas, 2022-08-17:
try this:
```
code: [-50000, 386, -1113, 392, -1121, 1110, -394, 1119, -389, 360, -1132, 368, -1148, 360, -1156, 1093, -385, 385, -1129, 369, -1124, 386, -1130, 369, -1125, 1111, -392, 1116, -388, 1112, -394, 368, -1141, 1111, -393, 369, -1144, 1090, -400, 374, -1143, 365, -1140, 1103, -399, 1102, -405, 1102, -400, 1101, -380, 375, -1145, 350, -1138, 375, -1144, 364, -50000]
```
I added a "-50000" at the beginning and end to ensure the radio leaves enough space between repetitions.
I suggest you do the repetitions through this instead: https://esphome.io/components/remote_transmitter.html?highlight=remote+transmitter#remote-transmitter-actions

ageurtse, 2022-08-17:
did recieve my second device but when trying to send data the other one isn't recieving the data, so looks like somthing is still wrong

dbuezas, 2022-08-17:
Did you try the alternative at the end of #2 ?

ageurtse, 2022-08-17:
i put your delay at the begin and at the end, but it still isn't working. 
don;t know what you mean with  [Did you try the alternative at the end of https://github.com/dbuezas/esphome-cc1101/issues/2 ?] 

if you mean, did you call endtransmission, than yes i did see my code. 

below is how i wired my device:
1	GND	GND		
2	VCC		VCC		
3	GDO0	D1		
4	CSN		D8	
5	SCK		D5		
6	MOSIO	D7		
7	MISO	D6		
8	GDO2	D2		

this is the device i have:[https://www.ebyte.com/en/downpdf.aspx?id=165]( https://www.ebyte.com/en/downpdf.aspx?id=165)

dbuezas, 2022-08-17:
Follow the link to the other issue here: #2 
This fellow used one pin for transmitting and another one for receiving, what makes the lambda calls unnecessary. He reported that that fixed some other issue I didn't have myself.

I see you already have GDO2 wired to d2, so you can use his modified yaml and cc1101.h files. He puts the chip in the right mode to use 2 wires. I did it with only 1 wire just because I have 3 chips and an IR receiver im the same board and ran out of pins :)

ageurtse, 2022-08-21:
still no good, could it be that the c1101 v2 is somewhat different. 
Today i tryed the Somfy code from here and that one also didn't work. 
[https://www.die-welt.net/2021/06/controlling-somfy-roller-shutters-using-an-esp32-and-esphome/](https://www.die-welt.net/2021/06/controlling-somfy-roller-shutters-using-an-esp32-and-esphome/)

i build this one upon an esp32 and like the schematic that's on that page. 

looks like this c1101 won't send data only recieve.

dbuezas, 2022-08-21:
That's really odd. The only way I know to know for sure is to get an rtl-sdr

ageurtse, 2022-08-21:
i will order some new ones at a different store with better reviews. 
after some digging i found out that there are a lot more problems with these modules. 

[https://www.amazon.nl/-/en/gp/product/B07YX92NMP/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1][(url)](https://www.amazon.nl/-/en/gp/product/B07YX92NMP/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

dbuezas, 2022-08-21:
I bought 3 different ones in AliExpress and all worked. I guess i got lucky then

ageurtse, 2022-08-24:
I also orderd 2 new ones from ali, lets hope they are working the way i want.

I have to wait till 20 september, so no more trying for me.

Keep you informed.

dbuezas, 2022-08-24:
No RTL-SDR? 😁

ageurtse, 2022-08-25:
No i dont't have 😣

dbuezas, 2022-08-25:
Ask santa to get you one. You'll be able to see the spectrum. They are very fun

ageurtse, 2022-09-08:
stil not working here, thinking of looking into a RTL-SDR. 
any advice on hardware and software. 
prefere free software and cheap hardware :-)

ageurtse, 2022-09-10:
have a RTL-SDR and reciving things now reinstall the esphome rf transmitter to see if it sends data

ageurtse, 2022-09-10:
it look like nothing is send, when pressing te remote there is data. 
when pressing the button within esphome nothing is displayed. 

looks like somthing is wrong, all connectings are right, the device is recieving data, but not sending. 
I tryed the default c1101.h and the alterd one where gdo2 is used.

dbuezas, 2022-09-10:
Can you post your yaml?
Also, do you have the cc1101.h file untouched as in this repo?

dbuezas, 2022-09-10:
Btw, you did confirm now with the rtl-sdr that the cc1101 is not sending, right?

ageurtse, 2022-09-10:
i confirmed that my c1101 isn't sending. 

below my cc1101.h file and yaml
from the yaml file, i only pasted the sensor section, the rest is for configuration. 

```#ifndef CC1101TRANSCIVER_H
#define CC1101TRANSCIVER_H

#include <ELECHOUSE_CC1101_SRC_DRV.h>

int CC1101_module_count = 0;
#define get_cc1101(id) (*((CC1101 *)id))

class CC1101 : public PollingComponent, public Sensor
{
  int _SCK;
  int _MISO;
  int _MOSI;
  int _CSN;
  int _GDO0;
  int _GDO2;
  float _bandwidth;

  float _moduleNumber;
  int _last_rssi = 0;

  void setup()
  {
    ELECHOUSE_cc1101.setGDO(_GDO0, _GDO2);
    ELECHOUSE_cc1101.addSpiPin(_SCK, _MISO, _MOSI, _CSN, _moduleNumber);
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setRxBW(_bandwidth);
    ELECHOUSE_cc1101.setMHZ(_freq);
    ELECHOUSE_cc1101.SetRx();
  }

public:
  float _freq;
  CC1101(int SCK, int MISO, int MOSI, int CSN, int GDO0, int GDO2, float bandwidth,
         float freq)
      : PollingComponent(100)
  {
    _SCK = SCK;
    _MISO = MISO;
    _MOSI = MOSI;
    _CSN = CSN;
    _GDO0 = GDO0;
    _GDO2 = GDO2;
    _bandwidth = bandwidth;
    _freq = freq;
    _moduleNumber = CC1101_module_count++;
  }

  void beginTransmission()
  {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.SetTx();
    noInterrupts();
  }
  void endTransmission()
  {
    interrupts();
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.setSidle();
  }
  void setRX()
  {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.SetRx();
  }
  void setBW(float bandwidth)
  {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.setRxBW(bandwidth);
  }
  void setFreq(float freq)
  {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.setMHZ(freq);
  }
  bool rssi_on;
  void update() override
  {
    int rssi = 0;
    if (rssi_on)
    {
      ELECHOUSE_cc1101.setModul(_moduleNumber);
      rssi = ELECHOUSE_cc1101.getRssi();
    }
    if (rssi != _last_rssi)
    {
      publish_state(rssi);
      _last_rssi = rssi;
    }
  }
};

#endif```

yaml
```
# --- Start of custom sensors ---

sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        D5, // SCK
        D6, // MISO
        D7, // MOSI
        D3, // CSN
        D1, // GDO0
        D2, // GDO2
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true

remote_transmitter:
  - pin: D2 # This is GDO2
    carrier_duty_percent: 100%

remote_receiver:
  - pin: D1 # This is GDO0
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump: all
      
button:
  - platform: template
    name: overkapping aan
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: '0011000100001110101001111000'
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();
      
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: '0011000100001110101001111000'
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();
      
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: '0011000100001110101001111000'
          protocol: 1
      - lambda: get_cc1101(transciver).endTransmission();
      
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_rc_switch_raw:
          code: '0011000100001110101001111000'
          protocol: 1
      - lambda: get_cc1101(transc
   ```

dbuezas, 2022-09-28:
Did you see this?
https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326/134

It seems like there is a bug in esphome regarding transmission, and the post also has a workaround :)

deanfourie1, 2023-05-20:
im having the same issues with the same module.

kgstorm, 2023-09-19:
One more data point: I have a cc1101 v2 as well, but a d1 mini, not lite. My cc1101 v2 DOES transmit with dbuezas’ yaml and .h file - although I struggled until getting an RTL-sdr.

kgstorm, 2023-09-19:
A year later so not sure if this is still an issue, but wondering if you gave your esphome component, looks like you named yours ‘118-Rf-tranceiver’ homeassistant access. 
That tripped me up for a while. Without permission to home assistant, when you press the button, the component won’t be notified.
To grant homeassistant permission, in home assistant, go to settings, devices & services, esphome, click ‘configure’ on your device, click ‘Allow the device to make Home Assistant service calls.’

deanfourie1, 2023-11-08:
Guys, im still fighting this.

I am able to receive, but I cannot send.
I have tried with dbuezas's yaml too.

Any ideas?

dbuezas, 2023-11-08:
@deanfourie1 
Do you have a second board or an stl-sdr to confirm that there is no signal being emitted?
It is likely that transmission works, but there is an issue with the code so your receiver device is not accepting it

deanfourie1, 2023-11-08:
> @deanfourie1 Do you have a second board or an stl-sdr to confirm that there is no signal being emitted? It is likely that transmission works, but there is an issue with the code so your receiver device is not accepting it

Sorry, I do not. This is literally my first time working with 433.

However, I am transmissing RAW, and on RX I am dumping RAW.

RX is showing some noise, so im sure the RX is receiving data.

For the TX, im really not sure. I have another board I can use to test. Im working off this pinout.

[image]

I also added.

```
          repeat: 
            times: 10
            wait_time: 0s
```

dbuezas, 2023-11-08:
@deanfourie1 what device are you trying to communicate with? The device may use a different style of communication (like modulating frequency) which will require a very different communication.

deanfourie1, 2023-11-08:
> Recommend adding a simple logger.log in the action to make sure those lines are being executed:
> 
> ```yaml
> button:
>   - platform: template
>     name: overkapping aan
>     on_press:
>       - logger.log: "Running button press"
> ```
> 
> If the logger.log statement doesn't print to the log, then likely the problem is that you didn't give your device permissions to access your home assistant instance.

Thanks I will do this.

deanfourie1, 2023-11-08:
> @deanfourie1 what device are you trying to communicate with? The device may use a different style of communication (like modulating frequency) which will require a very different communication.

CC1101 to CC1101

deanfourie1, 2023-11-08:
With the above added.

```
    on_press:
      - logger.log: "Running button press"
```

```
[22:48:04][D][button:010]: 'Garage' Pressed.
[22:48:04][D][remote_transmitter:075]: Sending remote code...
[22:48:04][D][main:084]: Running button press
[22:48:04][W][component:214]: Component api took a long time for an operation (0.05 s).
[22:48:04][W][component:215]: Components should block for at most 20-30ms.
[22:48:17][D][button:010]: 'Garage' Pressed.
[22:48:17][D][remote_transmitter:075]: Sending remote code...
[22:48:17][D][main:084]: Running button press
[22:48:17][W][component:214]: Component api took a long time for an operation (0.06 s).
[22:48:17][W][component:215]: Components should block for at most 20-30ms.
[22:48:20][D][button:010]: 'Garage' Pressed.
[22:48:20][D][remote_transmitter:075]: Sending remote code...
[22:48:20][D][main:084]: Running button press
[22:48:20][W][component:214]: Component api took a long time for an operation (0.05 s).
[22:48:20][W][component:215]: Components should block for at most 20-30ms.
```

kgstorm, 2023-11-08:
are you confident you have the right frequency? does the remote you are trying to reproduce have an FCC ID?

dbuezas, 2023-11-08:
> CC1101 to CC1101
So you do have a 2nd board?

deanfourie1, 2023-11-08:
> > CC1101 to CC1101
> 
> So you do have a 2nd board?

Yes correct, I actually have 4.

deanfourie1, 2023-11-08:
Receiver YAML

```
sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        D5, // SCK
        D6, // MISO
        D7, // MOSI
        D3, // CSN
        D1, // GDO0
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
#remote_transmitter:
#  - pin: D1 # This is GDO0
#    carrier_duty_percent: 100%

remote_receiver:
  - pin: D1 # This is GDO0
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump:
      - raw
binary_sensor:
  - platform: remote_receiver
    name: Garage
    raw:
      code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
#button:
#  - platform: template
#    name: Garage
#    on_press:
#      - lambda: get_cc1101(transciver).beginTransmission();
#      - remote_transmitter.transmit_raw:
#          code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
#          repeat: 
#            times: 10
#            wait_time: 0s
#      - lambda: get_cc1101(transciver).endTransmission();
```

Transmitter YAML

```
sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        D5, // SCK
        D6, // MISO
        D7, // MOSI
        D3, // CSN
        D1, // GDO0
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: D1 # This is GDO0
    carrier_duty_percent: 100%

#remote_receiver:
#  - pin: D1 # This is GDO0
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
#    dump:
#      - raw
#binary_sensor:
#  - platform: remote_receiver
#    name: Garage
#    raw:
#      code: code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
#          repeat: 
#            times: 10
#            wait_time: 0s
      - lambda: get_cc1101(transciver).endTransmission();
      - logger.log: "Running button press"
```

kgstorm, 2023-11-08:
that is a pretty short code...
Can you send a log of what is received when you hit the remote button?

deanfourie1, 2023-11-08:
[image]

[image]

[image]

[image]

deanfourie1, 2023-11-08:
> that is a pretty short code... Can you send a log of what is received when you hit the remote button?

Yea sure, looks to be sending very random data.

```
[23:06:46][I][remote.raw:041]: Received Raw: 418, -293, 384, -283, 390, -301, 374, -297, 382, -286, 420
[23:06:50][I][remote.raw:041]: Received Raw: 389, -305, 375, -288, 371, -295, 383, -306, 399, -300, 400
[23:06:51][I][remote.raw:041]: Received Raw: 428, -250, 395, -310, 370, -301, 374, -298, 365, -311, 396
[23:06:52][I][remote.raw:041]: Received Raw: 418, -272, 357, -310, 394, -271, 403, -307, 372, -306, 395
[23:06:53][I][remote.raw:041]: Received Raw: 413, -279, 388, -311, 366, -295, 382, -302, 365, -306, 395
[23:06:53][I][remote.raw:041]: Received Raw: 388, -299, 376, -300, 379, -311, 369, -296, 381, -329, 394
[23:06:54][I][remote.raw:041]: Received Raw: 386, -284, 375, -286, 397, -320, 358, -311, 369, -300, 402
[23:06:54][I][remote.raw:041]: Received Raw: 418, -270, 403, -303, 351, -312, 371, -295, 383, -305, 399
[23:06:55][I][remote.raw:041]: Received Raw: 419, -275, 396, -299, 389, -287, 372, -294, 384, -306, 399
[23:06:55][I][remote.raw:041]: Received Raw: 408, -276, 388, -287, 371, -317, 360, -308, 370, -306, 394
[23:06:56][I][remote.raw:041]: Received Raw: 382, -301, 378, -284, 390, -295, 383, -308, 362, -304, 396
[23:06:56][I][remote.raw:041]: Received Raw: 417, -289, 392, -277, 396, -295, 358, -310, 370, -299, 402
[23:06:56][I][remote.raw:041]: Received Raw: 411, -276, 358, -311, 394, -295, 383, -305, 373, -311, 397
[23:06:57][I][remote.raw:041]: Received Raw: 412, -280, 355, -310, 394, -321, 357, -307, 372, -311, 399
[23:06:57][I][remote.raw:041]: Received Raw: 431, -277, 389, -284, 392, -293, 384, -282, 390, -277, 422
[23:06:57][I][remote.raw:041]: Received Raw: 416, -285, 395, -297, 380, -302, 351, -311, 371, -296, 407
[23:06:57][I][remote.raw:041]: Received Raw: 419, -287, 370, -299, 377, -300, 378, -312, 370, -295, 407
[23:06:58][I][remote.raw:041]: Received Raw: 405, -297, 381, -311, 369, -297, 382, -300, 351, -337, 400
[23:06:58][I][remote.raw:041]: Received Raw: 416, -286, 371, -319, 359, -307, 372, -306, 370, -295, 407
[23:06:58][I][remote.raw:041]: Received Raw: 391, -281, 383, -304, 371, -294, 407, -309, 367, -295, 408
[23:06:58][I][remote.raw:041]: Received Raw: 403, -299, 380, -311, 369, -296, 381, -303, 350, -311, 397
[23:06:58][I][remote.raw:041]: Received Raw: 390, -308, 367, -295, 383, -305, 365, -305, 371, -319, 383
[23:06:59][I][remote.raw:041]: Received Raw: 420, -294, 381, -303, 375, -286, 372, -295, 383, -307, 399
[23:06:59][I][remote.raw:041]: Received Raw: 417, -294, 380, -305, 374, -287, 371, -319, 360, -306, 397
```

deanfourie1, 2023-11-08:
Units are very close together, not sure if that matters.

deanfourie1, 2023-11-08:
Any idea what could cause sending of absolute random codes?

kgstorm, 2023-11-08:
It doesn't look random. It looks very consistent. Those numbers are nanoseconds, so the measured duration fluctuating by +/- 20 nanoseconds is to be expected.
try adding some delay to your repeat. maybe 10ms
```yaml
          repeat: 
            times: 10
            wait_time: 10ms
```

deanfourie1, 2023-11-08:
> ```yaml
>           repeat: 
>             times: 10
>             wait_time: 10ms
> ```

Oh ok, however the strange thing is it is not triggering the garage binary sensor.

deanfourie1, 2023-11-08:
Oh ok you legend! I think thats working.

It turns on and off so damn fast it almost hard to see.

deanfourie1, 2023-11-08:
Is it possible to make it "latching" until a restore is sent. Like a door.

dbuezas, 2023-11-08:
The data you are receiving looks almost perfect.
The only issue is that it is receiving an extra pulse (probably because of the 0 wait you put there (as @kgstorm mentioned).

[image]

These numbers represent the time between each flank of each pulse. They only need to be maybe ±10%

deanfourie1, 2023-11-08:
> The data you are receiving looks almost perfect. The only issue is that it is receiving an extra pulse (probably because of the 0 wait you put there (as @kgstorm mentioned).
> 
> [image]
> (last row is the sent code)
> 
> These numbers represent the time between each flank of each pulse. They only need to be maybe ±10%

Oh wow, that explains it well.

I was expecting it to receive the exact same data as being sent, but its obviously encoded somehow right.

dbuezas, 2023-11-08:
BTW, here's the online web plotting tool: https://github.com/dbuezas/esphome-remote_receiver-oscilloscope

deanfourie1, 2023-11-08:
> BTW, here's the online web plotting tool: https://github.com/dbuezas/esphome-remote_receiver-oscilloscope

Ive actually come accross that before, but the data I put in looked nothing like that haha. I think it was just a lot of noise.

deanfourie1, 2023-11-08:
Thanks for your help guys! I'm super wrapped to finally get this working!

deanfourie1, 2023-11-08:
> > Is it possible to make it "latching" until a restore is sent. Like a door.
> 
> ```yaml
> binary_sensor:
>   - platform: template
>     name: "Door Keller"
>     id: door_keller
>     device_class: door
>   - platform: remote_receiver
>     receiver_id: receiver_doors
>     id: keller_door_off
>     internal: true
>     name: ""
>     on_press:
>       - binary_sensor.template.publish:
>           id: door_keller
>           state: ON
>     raw:
>       code:
>         [.....]
>   - platform: remote_receiver
>     receiver_id: receiver_doors
>     id: keller_door_on
>     internal: true
>     name: ""
>     on_press:
>       - binary_sensor.template.publish:
>           id: door_keller
>           state: OFF
>     raw:
>       code:
>         [.....]
>   
> ```

Thanks, I need to trigger this from a GPIO.

Another question, how can I generate the [code] to be sent. Do I just make it up?

dbuezas, 2023-11-08:
> Thanks, I need to trigger this from a GPIO.

Look into esphome docs you can do anything with it.

>  Another question, how can I generate the [code] to be sent. Do I just make it up?

That's exactly the raw output you get in the receiver

deanfourie1, 2023-11-08:
But what I dont understand is, the commands sent by transmitter were different to what was received by the receiver.

dbuezas, 2023-11-08:
Mmm were they? Can you run this again and post the result after adding the delay between repeats?
Pleas also post what you transmit

deanfourie1, 2023-11-08:
Hey could you help me out with one more issue. Im using the Wemos Battery Shield which is a WROOM 02 ESP.

https://epartners.co.nz/products/tm2004?_pos=3&_sid=e0721fd63&_ss=r

When compiling the full code, the device fails to boot or connect to HA.

I'm thinking this could be due to me using a strapping pin defined in the 

```
sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        5, // SCK
        6, // MISO
        7, // MOSI
        3, // CSN
        1, // GDO0
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: 1 # This is GDO0
    carrier_duty_percent: 100%
```

Do you have any idea how I can get around this. Full YAML.

```
esphome:
  name:
  friendly_name:
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"

esp8266:
  board: esp_wroom_02

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: 
    password: 

captive_portal:
    
sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        5, // SCK
        6, // MISO
        7, // MOSI
        3, // CSN
        1, // GDO0
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: 1 # This is GDO0
    carrier_duty_percent: 100%

button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
          repeat: 
            times: 10
            wait_time: 10ms
      - lambda: get_cc1101(transciver).endTransmission();
```

Thank you!

deanfourie1, 2023-12-20:
Hey guys, is there any way we can get the CC1101 to send the commands from this post?

Much appreciated!

https://github.com/esphome/feature-requests/issues/2235#issuecomment-1858834643

[1 more reply]

closed 2024-06-08 by dbuezas