dbuezas/esphome-cc1101 #7: Not able to send or receive

deanfourie1, 2023-05-20:
Guys, I am unable to get this module to send / receive. I guess receive is not working because send is failing.

Here my setup. I have 2x ESP8266 WeeMos D1 minis setup, with 2 of these modules connect to each esp via the wiring outlined in the Repo.

https://amptech.co.nz/CC1101-Wireless-Transceiver-Module-SMA-Antenna-Arduino-433MHZ?search=cc1101

I have added the demo yaml configuration to both devices in ESPhome and installed on the esp.

If I go one of the ESPs and open the Garage binary sensor, and then open the other ESP and open the Garage button, when I press the button I do not see the Garage binary sensor changing state.

Any ideas on this one? It is all untouched config straight from the REPO. I have everything wired execpt for GDO2.

Thanks

dbuezas, 2023-05-20:
I haven't updated it's firmware in a while, maybe the remote receiver component in EspHome has changed.
The one I'm running had 2023.3.0 installed

deanfourie1, 2023-05-20:
Its weird though, because if I run a log output of the receiving ESP I get a RAW code.

Its spits out a random value, a 4 digit code like 4234 or whatever.

Is this expected behaviour.

However, still no activation on the Output

[image]

deanfourie1, 2023-05-20:
Transmit and receive logs.

`[23:43:09][D][button:010]: 'Garage' Pressed.
[23:43:09][D][remote_transmitter:075]: Sending remote code...
[23:43:10][D][button:010]: 'Garage' Pressed.
[23:43:10][D][remote_transmitter:075]: Sending remote code...
[23:43:11][D][button:010]: 'Garage' Pressed.
[23:43:11][D][remote_transmitter:075]: Sending remote code...`

`[23:50:29][D][remote.raw:041]: Received Raw: 4261
[23:50:30][D][remote.raw:041]: Received Raw: 4236
[23:50:30][D][remote.raw:041]: Received Raw: 4238
[23:50:30][D][remote.raw:041]: Received Raw: 4220
[23:50:30][D][remote.raw:041]: Received Raw: 4221
[23:50:30][D][remote.raw:041]: Received Raw: 4217
[23:50:30][D][remote.raw:041]: Received Raw: 4220
[23:50:31][D][remote.raw:041]: Received Raw: 4203
[23:50:31][D][remote.raw:041]: Received Raw: 4247
[23:50:32][D][remote.raw:041]: Received Raw: 4230
[23:50:32][D][remote.raw:041]: Received Raw: 4246
[23:50:32][D][remote.raw:041]: Received Raw: 4222
[23:50:32][D][remote.raw:041]: Received Raw: 4214
[23:50:33][D][remote.raw:041]: Received Raw: 4238`

deanfourie1, 2023-05-20:
> 

Just any remote? Or use the ESP transmitter with the Garage button to send command?

dbuezas, 2023-05-20:
Any remote that emits in the right frequency

dbuezas, 2023-05-20:
The rationale is that it may be that something goes wrong after transmitting from the cc1101 for the first time. If it works until you transmit, it will be a good cue about what's going on

deanfourie1, 2023-05-21:
Ok I just did this, 

Pushed a button on another 433 transmitter and this is what I got.

`[17:44:18][D][remote.raw:041]: Received Raw: 
[17:44:18][D][remote.raw:041]: Received Raw: 
[17:44:18][D][remote.raw:041]: Received Raw: -2534
[17:44:18][D][remote.raw:041]: Received Raw: -3722
[17:44:18][D][remote.raw:041]: Received Raw: -3997
[17:44:18][D][remote.raw:041]: Received Raw: -3947
[17:44:18][D][remote.raw:041]: Received Raw: -3979
[17:44:18][D][remote.raw:041]: Received Raw: -3941
[17:44:18][D][remote.raw:041]: Received Raw: -3961
[17:44:18][D][remote.raw:041]: Received Raw: -3980
[17:44:18][D][remote.raw:041]: Received Raw: -3976
[17:44:18][D][remote.raw:041]: Received Raw: -3946
[17:44:19][D][remote.raw:041]: Received Raw: -3975
[17:44:19][D][remote.raw:041]: Received Raw: -3972
[17:44:19][D][remote.raw:041]: Received Raw: -3980
[17:44:19][D][remote.raw:041]: Received Raw: 
[17:44:19][D][remote.raw:041]: Received Raw: 
[17:44:19][D][remote.raw:041]: Received Raw: -3625
[17:44:19][D][remote.raw:041]: Received Raw: -3943
[17:44:19][D][remote.raw:041]: Received Raw: -3994
[17:44:19][D][remote.raw:041]: Received Raw: -3977
[17:44:19][D][remote.raw:041]: Received Raw: -3974
[17:44:19][D][remote.raw:041]: Received Raw: -3970
[17:44:19][D][remote.raw:041]: Received Raw: -3962
[17:44:19][D][remote.raw:041]: Received Raw: -3972
[17:44:19][D][remote.raw:041]: Received Raw: -3980
[17:44:19][D][remote.raw:041]: Received Raw: -3972
[17:44:19][D][remote.raw:041]: Received Raw: -3994
[17:44:19][D][remote.raw:041]: Received Raw: -3937
[17:44:19][D][remote.raw:041]: Received Raw: -3986
[17:44:19][D][remote.raw:041]: Received Raw: -3978
[17:44:19][D][remote.raw:041]: Received Raw: 
[17:44:19][D][remote.raw:041]: Received Raw: 
[17:44:19][D][remote.raw:041]: Received Raw: -3596
[17:44:19][D][remote.raw:041]: Received Raw: -3974
[17:44:19][D][remote.raw:041]: Received Raw: -3969
[17:44:19][D][remote.raw:041]: Received Raw: -3974
[17:44:19][D][remote.raw:041]: Received Raw: -3954
[17:44:19][D][remote.raw:041]: Received Raw: -3977
[17:44:19][D][remote.raw:041]: Received Raw: -3984
[17:44:19][D][remote.raw:041]: Received Raw: -3973
[17:44:19][D][remote.raw:041]: Received Raw: -3971
[17:44:19][D][remote.raw:041]: Received Raw: -3980
[17:44:19][D][remote.raw:041]: Received Raw: -3979
[17:44:19][D][remote.raw:041]: Received Raw: -3968
[17:44:19][D][remote.raw:041]: Received Raw: -3972
[17:44:19][D][remote.raw:041]: Received Raw: -3954
[17:44:20][D][remote.raw:041]: Received Raw: 
[17:44:20][D][remote.raw:041]: Received Raw: 
[17:44:20][D][remote.raw:041]: Received Raw: -4400
[17:44:20][D][remote.raw:041]: Received Raw: -4414
[17:44:20][D][remote.raw:041]: Received Raw: -4424
[17:44:20][D][remote.raw:041]: Received Raw: -4269
[17:44:20][D][remote.raw:041]: Received Raw: -4230
[17:44:20][D][remote.raw:041]: Received Raw: -4098
[17:44:20][D][remote.raw:041]: Received Raw: -4378
[17:44:20][D][remote.raw:041]: Received Raw: -4376
[17:44:20][D][remote.raw:041]: Received Raw: -4388
[17:44:20][D][remote.raw:041]: Received Raw: -4400
[17:44:20][D][remote.raw:041]: Received Raw: -4157
[17:44:20][D][remote.raw:041]: Received Raw: -3721
[17:44:21][D][remote.raw:041]: Received Raw: -4128
[17:44:24][D][remote.raw:041]: Received Raw: -4088
[17:44:24][D][remote.raw:041]: Received Raw: 
[17:44:24][D][remote.raw:041]: Received Raw: 
[17:44:24][D][remote.raw:041]: Received Raw: -4260
[17:44:24][D][remote.raw:041]: Received Raw: -4127
[17:44:24][D][remote.raw:041]: Received Raw: -7255
[17:44:24][D][remote.raw:041]: Received Raw: -4385
[17:44:24][D][remote.raw:041]: Received Raw: -4359
[17:44:24][D][remote.raw:041]: Received Raw: -4398
[17:44:24][D][remote.raw:041]: Received Raw: -4405
[17:44:24][D][remote.raw:041]: Received Raw: -4271
[17:44:24][D][remote.raw:041]: Received Raw: -4208
[17:44:24][D][remote.raw:041]: Received Raw: -4348
[17:44:24][D][remote.raw:041]: Received Raw: -4119
[17:44:24][D][remote.raw:041]: Received Raw: -4441
[17:44:24][D][remote.raw:041]: Received Raw: -7760
[17:44:24][D][remote.raw:041]: Received Raw: -4342
[17:44:24][D][remote.raw:041]: Received Raw: 
[17:44:24][D][remote.raw:041]: Received Raw: 
[17:44:24][D][remote.raw:041]: Received Raw: -4346
[17:44:24][D][remote.raw:041]: Received Raw: -4426
[17:44:24][D][remote.raw:041]: Received Raw: -4346
[17:44:24][D][remote.raw:041]: Received Raw: -4135
[17:44:24][D][remote.raw:041]: Received Raw: -4153
[17:44:24][D][remote.raw:041]: Received Raw: -4126
[17:44:24][D][remote.raw:041]: Received Raw: -4221
[17:44:24][D][remote.raw:041]: Received Raw: -4141
[17:44:24][D][remote.raw:041]: Received Raw: -4096
[17:44:24][D][remote.raw:041]: Received Raw: -4333
[17:44:24][D][remote.raw:041]: Received Raw: -4373
[17:44:24][D][remote.raw:041]: Received Raw: -4378
[17:44:24][D][remote.raw:041]: Received Raw: -4149
[17:44:24][D][remote.raw:041]: Received Raw: -4409`

dbuezas, 2023-05-21:
That doesn't look quite right. It should vte many numbers per line alternating positive and negative.

deanfourie1, 2023-05-21:
What do you think it could be?

dbuezas, 2023-05-21:
Hard to say, mine is working fine with esphome 2023.3.
You didn't change anything from the yaml or h files, right?
Are you using the exact same gpio pins?

dbuezas, 2023-05-21:
Also, are you 100% sure your remote uses 433.92MHz?

deanfourie1, 2023-05-21:
> Hard to say, mine is working fine with esphome 2023.3. You didn't change anything from the yaml or h files, right? Are you using the exact same gpio pins?

The remote used was 433mhz bang on.

Wait, I used the other version of the library. Not the one on here as I could not get anything to transmit.

I have wired my GDO2 to ping D2 on the ESP as I couldn't get it to transmit.

Following the other issue, as described there.

dbuezas, 2023-05-21:
Yes, there seem to be an incompatibility with esp32 and this custom component. But not with esp8266

deanfourie1, 2023-05-21:
Oh ok, well im using a ESP8266, wemos D1 mini to be exact.

Should I try switch back to your library?

deanfourie1, 2023-05-21:
What is the best way to replace the library? Just overwrite the .h file? 

Need to recompile and reinstall?

Thanks

deanfourie1, 2023-05-21:
Full log of compilation

`INFO Reading configuration /config/esphome/rf-receiver.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing rf-receiver (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Library Manager: Installing SmartRC-CC1101-Driver-Lib

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    2%
Unpacking  [#-----------------------------------]    4%
Unpacking  [##----------------------------------]    6%
Unpacking  [###---------------------------------]    8%
Unpacking  [####--------------------------------]   11%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   15%
Unpacking  [######------------------------------]   17%
Unpacking  [#######-----------------------------]   20%
Unpacking  [########----------------------------]   22%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   26%
Unpacking  [##########--------------------------]   28%
Unpacking  [###########-------------------------]   31%
Unpacking  [############------------------------]   33%
Unpacking  [############------------------------]   35%
Unpacking  [#############-----------------------]   37%
Unpacking  [##############----------------------]   40%
Unpacking  [###############---------------------]   42%
Unpacking  [################--------------------]   44%
Unpacking  [################--------------------]   46%
Unpacking  [#################-------------------]   48%
Unpacking  [##################------------------]   51%
Unpacking  [###################-----------------]   53%
Unpacking  [####################----------------]   55%
Unpacking  [####################----------------]   57%
Unpacking  [#####################---------------]   60%
Unpacking  [######################--------------]   62%
Unpacking  [#######################-------------]   64%
Unpacking  [########################------------]   66%
Unpacking  [########################------------]   68%
Unpacking  [#########################-----------]   71%
Unpacking  [##########################----------]   73%
Unpacking  [###########################---------]   75%
Unpacking  [############################--------]   77%
Unpacking  [############################--------]   80%
Unpacking  [#############################-------]   82%
Unpacking  [##############################------]   84%
Unpacking  [###############################-----]   86%
Unpacking  [################################----]   88%
Unpacking  [################################----]   91%
Unpacking  [#################################---]   93%
Unpacking  [##################################--]   95%
Unpacking  [###################################-]   97%
Unpacking  [####################################]  100%
Library Manager: SmartRC-CC1101-Driver-Lib@2.5.7 has been installed!
Dependency Graph
|-- SPI @ 1.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/api_connection.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/api_server.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/list_entities.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/proto.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/src/esphome/components/api/subscribe_state.cpp.o
[... 179 lines cut ...]
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/mmu_iram.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/reboot_uart_dwnld.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/spiffs/spiffs_cache.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/spiffs_api.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/sqrt32.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/stdlib_noniso.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/time.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/uart.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling /data/rf-receiver/.pioenvs/rf-receiver/FrameworkArduino/umm_malloc/umm_poison.c.o
[... cut ...]
Linking /data/rf-receiver/.pioenvs/rf-receiver/firmware.elf
RAM:   [====      ]  38.2% (used 31304 bytes from 81920 bytes)
Flash: [====      ]  35.9% (used 374783 bytes from 1044464 bytes)
Building /data/rf-receiver/.pioenvs/rf-receiver/firmware.bin
esp8266_copy_factory_bin(["/data/rf-receiver/.pioenvs/rf-receiver/firmware.bin"], ["/data/rf-receiver/.pioenvs/rf-receiver/firmware.elf"])
========================= [SUCCESS] Took 38.87 seconds =========================
INFO Successfully compiled program.`

deanfourie1, 2023-05-21:
Ok here's an update!

I redid everything using your original library, and straight aware got it working without connecting GDO2 to D2.

Upon sending the codes, the correct codes are received however the output still did not turn on.

I pressed the button repetitively about 10 - 20 times quickly and eventually the output switched to the on state.

So that confirms it is working,

Is there a filter I can use to send the command x amount of times to ensure it registers?

Thanks, sorry for wasting your time with using the other library.

dbuezas, 2023-05-21:
Oh, you weren't using what's in this repo! I missed that! 😆. 

To repeat add this to your transmit part
```
on_...:
  - remote_transmitter.transmit_pioneer:
      rc_code_1: 0xA556
      rc_code_2: 0xA506
      repeat:
        times: 2
```

I strongly recommend you read the docs here: https://esphome.io/components/remote_transmitter.html#remote-transmitter-actions

I think you got far enough to understand everything there :)

closed 2023-05-21 by dbuezas

deanfourie1, 2023-05-21:
Just one thing, its still spitting out random values.

Is that normal?

dbuezas, 2023-05-21:
BTW, the link you put in the first post gives me a 403 forbidden. It looks like [https://amptech.co.nz/] is not accessible from where I live

dbuezas, 2023-05-21:
> Just one thing, its still spitting out random values.
> 
> Is that normal?

Yes, that is normal, particularly with raw patterns.

They are just interference. The CC1101 automatically tunes its gain so if there are no signals in the air, it starts picking up noise (and/or other transmitters).

deanfourie1, 2023-05-21:
ah ok try this one.

https://www.amazon.com/CC1101-Wireless-Transceiver-915MHZ-Antenna/dp/B01DS1WUEQ

deanfourie1, 2023-05-21:
> > Just one thing, its still spitting out random values.
> > Is that normal?
> 
> Yes, that is normal, particularly with raw patterns.
> 
> They are just interference. The CC1101 automatically tunes its gain so if there are no signals in the air, it starts picking up noise (and/or other transmitters).

ah ok in that case I guess its working.

Thank you for your help and for creating this!!

dbuezas, 2023-05-21:
> ah ok try this one.
> 
> https://www.amazon.com/CC1101-Wireless-Transceiver-915MHZ-Antenna/dp/B01DS1WUEQ

Oh, that's just the board. I thought you had some other library.
Btw, these are WAY cheaper in ali express

deanfourie1, 2023-05-21:
Yea I bought local anyway.

I was referring to library here.

https://github.com/dbuezas/esphome-cc1101/issues/2#issuecomment-1172924868