dbuezas/esphome-cc1101 #8: cannot transmit or receive

deanfourie1, 2023-05-27:
I am still unable to get this to work.

I've shifted over to a Wemos ESP WROOM02 D1 Battery Shield and using the same code, I get a log message saying that the remote code was sent but nothing received.

On my receiver, I do seem to be getting a lot of noise, so I take it its receiving ok?? But still, no sign of the code received and no trigger on the garage output.

One other thing, with the stock YAML, same code but one on a WemosD1 mini and one on the Wemos D1 Battery Sheild, however I do not get any of the noise on the WROOM02 shield as I do on the D1mini.

Any ideas?
Thanks

Pic of the noise I receive

[image]

dbuezas, 2023-05-27:
do you have a the frequency setup, on a matching antenna and board? 
there are 2 versions of the cc1101 boards: one is optimized for 868 and the other to 432 Mhz.

Regarding the noise, it could be that the battery shield has less noise on 3.3v rail and that that results in less noise (maybe)

Are you using the exact same gpios as in the yaml in this repo? not all gpios are created equal in the esp8266

deanfourie1, 2023-05-27:
Thanks for the reply.

Yes, It has D1 to D8 labelled. So wired identically to the D1mini.

I bought 4 of these units, so I have 2 of them setup, with the same antennas etc, one wired to the D1 battery Shield and one to the D1 mini.

Thanks

deanfourie1, 2023-05-27:
This is the guy I got so i'm guessing 433.

[image]

dbuezas, 2023-05-27:
Can you double check that your YAML matches all pins from here: https://github.com/dbuezas/esphome-cc1101/blob/main/cc1101.yaml

And also that your connections are on the right pins in the cc1101 and esp board

deanfourie1, 2023-05-27:
to be fair, i have tripple checked. I'll have another look but 99% sure its not wiring,

Thanks

dbuezas, 2023-05-27:
Worth also checking that vcc and gnd weren't forgotten,  or you accidentally looked at a pinout from below or something silly like that. I've lost weeks with simple issues of this nature: 😅

deanfourie1, 2023-05-28:
Ok i've gone over wiring again tonight, it's definitely not the wiring.

dbuezas, 2023-05-28:
Then I'm out of ideas. Pictures?

deanfourie1, 2023-05-28:
Sure, here are pics of both transmitter and receiver.

Let me know if you want closer shots.

[image]

[image]

closed 2024-06-08 by dbuezas