dbuezas/esphome-cc1101 #10: Wiring E07-M1101S (C1101) with D1 mini

flouck, 2023-07-17:
Hey,
Thanks for the code! I hope it will help me characterize my store remote signals to reproduce them as I want.
Could you please guide me into the wiring diagram between the two components?
I have two small red electronic cards so I guess one is the emitter and the other one is the receiver, right?
Is antenna mandatory?
Thanks for your help 😊

dbuezas, 2023-07-17:
Hi!
For the wiring google for the pinout of the board, then the names should match the ones in the code.
The antenna is necessary, otherwise your range will be very poor.
You can use a single board as both receiver and transmitter, checkout the yaml files :)

dbuezas, 2023-07-17:
You can leave GDO2 disconnected. For ease of use, the approach here is to configure the c1101 to both send and receive through gdo0

jbrandek, 2023-10-11:
sorry for my probably dumb question but it woludnt be better to have GDO0 to send and GDO2 to receive?

dbuezas, 2023-10-11:
Probably. I made this for my gadget that uses 3 transcivers, and an IR receiver in an esp8266 and I ran out of pins. And memory.

dbuezas, 2023-10-11:
Yes. You can still not receive while transmitting because it is a single antenna, and the transmitter would anyway completely saturate the receiver anyway.
You also need to change the custom component file (c1101.h) to configure the module to use both pins. Check out the docs of the used library for that.

closed 2024-06-08 by dbuezas