dbuezas/esphome-cc1101 #13: Grothe Mistral SE doorbell button

jwaes, 2023-07-26:
Sorry to abuse the 'issues' for a support question.

I have a doorbell button like this (https://img.ricardostatic.ch/images/f764d4ee-3e9c-4652-af2f-382893df9216/t_1800x1350/grothe-sender-mistral-se-011b)
It mentions 868.35Mhz

and i also found this code on github (https://raw.githubusercontent.com/Ralf9/RFFHEM/master/FHEM/lib/signalduino_protocols.pm)
that is about the same device with hopefully some usefull info:

```perl
		"96"	=>	# Funk-Gong | Taster Grothe Mistral SE 03.1 / 01.1, Innenteil Grothe Mistral 200M(E)
							# https://forum.fhem.de/index.php/topic,64251.msg940593.html?PHPSESSID=nufcvvjobdd8r7rgr0cq3qkrv0#msg940593 @coolheizer
							# SD_BELL_104762 Alarm        MC;LL=-430;LH=418;SL=-216;SH=226;D=23C823B1401F8;C=214;L=49;R=53;
							# SD_BELL_104762 ring         MC;LL=-439;LH=419;SL=-221;SH=212;D=238823B1001F8;C=215;L=49;R=69;
							# SD_BELL_104762 ring low bat MC;LL=-433;LH=424;SL=-214;SH=210;D=238823B100248;C=213;L=49;R=65;
							# SD_BELL_0253B3 Alarm        MC;LL=-407;LH=451;SL=-195;SH=239;D=23C129D9E78;C=215;L=41;R=241;
							# SD_BELL_0253B3 ring         MC;LL=-412;LH=458;SL=-187;SH=240;D=238129D9A78;C=216;L=41;R=241;
							# SD_BELL_024DB5 Alarm        MC;LL=-415;LH=454;SL=-200;SH=226;D=23C126DAE58;C=215;L=41;R=246;
							# SD_BELL_024DB5 ring         MC;LL=-409;LH=448;SL=-172;SH=262;D=238126DAA58;C=215;L=41;R=238;
			{
				name            => 'Grothe Mistral SE',
				comment         => 'Wireless doorbell Grothe Mistral SE 01.1 or 03.1',
				changed         => '20190518 new',
				id              => '96',
				knownFreqs      => '868.35',
				clockrange      => [170,260],
				format          => 'manchester',
				clientmodule    => 'SD_BELL',
				modulematch     => '^P96#',
				preamble        => 'P96#',
				length_min      => '40',
				length_max      => '49',
				method          => \&main::SIGNALduino_GROTHE,		# Call to process this message
			},
```

I started with the example yaml:

```yaml
esphome:
  name: buzzer
  friendly_name: buzzer
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"

esp8266:
  board: d1_mini

[...]
    
sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        D5, // SCK
        D6, // MISO
        D7, // MOSI
        D3, // CSN
        D1, // GDO0
        200, // bandwidth_in_khz
        868.35 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: D1 # This is GDO0
    carrier_duty_percent: 100%

remote_receiver:
  - pin: D1 # This is GDO0
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump:
      - raw
      
binary_sensor:
  - platform: remote_receiver
    name: Garage
    raw:
      code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [415,-300,370,-300,375,-300,378,-300,378,-300,379,-300,405]
      - lambda: get_cc1101(transciver).endTransmission();
```

I do not get any errors, but i was expecting 'something' to show up in the logs from the `remote_receiver` if i pushed the doorbell button.

The garage button seems to trigger the transmitter:

```
[18:14:37][C][logger:301]: Logger:
[18:14:37][C][logger:302]:   Level: DEBUG
[18:14:37][C][logger:303]:   Log Baud Rate: 115200
[18:14:37][C][logger:305]:   Hardware UART: UART0
[18:14:37][C][remote_transmitter:018]: Remote Transmitter...
[18:14:37][C][remote_transmitter:019]:   Carrier Duty: 100%
[18:14:37][C][remote_transmitter:020]:   Pin: GPIO5
[18:14:37][C][remote_base:027]: Remote Receiver Binary Sensor 'Garage'
[18:14:37][C][remote_receiver.esp8266:060]: Remote Receiver:
[18:14:37][C][remote_receiver.esp8266:061]:   Pin: GPIO5
[18:14:37][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[18:14:37][C][remote_receiver.esp8266:067]:   Tolerance: 25%
[18:14:37][C][remote_receiver.esp8266:068]:   Filter out pulses shorter than: 50 us
[18:14:37][C][remote_receiver.esp8266:069]:   Signal is done after 10000 us of no changes
[18:14:37][C][captive_portal:088]: Captive Portal:
[18:14:37][C][mdns:112]: mDNS:
[18:14:37][C][mdns:113]:   Hostname: buzzer
[18:14:37][C][ota:093]: Over-The-Air Updates:
[18:14:37][C][ota:094]:   Address: buzzer.local:8266
[18:14:37][C][ota:097]:   Using Password.
[18:14:37][C][api:138]: API Server:
[18:14:37][C][api:139]:   Address: buzzer.local:6053
[18:14:37][C][api:141]:   Using noise encryption: YES
[18:14:55][D][button:010]: 'Garage' Pressed.
[18:14:55][D][remote_transmitter:075]: Sending remote code...
```

I guess if i hooked the cc1101 board incorrectly i would get explicit errors ?

jwaes, 2023-07-26:
Okay ....
it looks like i *did* switch up some wires.

This already looks more like 'something'

```
[21:03:50][D][remote.raw:041]: Received Raw: 
[21:03:50][D][remote.raw:028]: Received Raw: 228, -436, 211, -240, 418, -440, 232, -213, 214, -215, 211, -239, 204, -215, 443, -433, 435, -236, 212, -431, 1521, -236, 212, -446, 424, -223, 215, -219, 229, -437, 210, -240, 215, -213, 436, -208, 238, -213, 213, -213, 231, -215, 218, 
[21:03:50][D][remote.raw:028]:   -428, 438, -236, 213, -446, 424, -223, 215, -220, 230, -234, 212, -213, 213, -444, 208, -240, 431, -206, 231, -214, 219, -429, 234, -213, 443, -440, 207, -214, 240, -214, 213, -214, 213, -213, 450, -421, 457, -210, 238, -418, 1542, -201, 230, -437, 
[21:03:50][D][remote.raw:028]:   435, -233, 191, -238, 216, -442, 233, -211, 193, -238, 437, -197, 242, -224, 206, -236, 193, -237, 219, -440, 441, -206, 237, -415, 440, -235, 212, -214, 240, -213, 213, -212, 231, -436, 228, -207, 438, -236, 190, -238, 216, -436, 228, -207, 438, -438, 
[21:03:50][D][remote.raw:028]:   236, -213, 214, -214, 214, -214, 213, -241, 427, -442, 428, -238, 206, -442, 1508, -225, 230, -436, 439, -210, 214, -239, 215, -441, 206, -237, 193, -213, 462, -198, 241, -223, 206, -236, 192, -238, 216, -435, 448, -206, 237, -415, 439, -223, 216, 
[21:03:50][D][remote.raw:028]:   -221, 229, -236, 212, -212, 213, -439, 235, -213, 430, -231, 204, -216, 220, -456, 210, -212, 443, -440, 206, -239, 212, -213, 213, -238, 205, -215, 442, -433, 439, -209, 239, -431, 1521, -236, 212, -444, 414, -232, 211, -231, 215, -443, 207, -237, 
[21:03:50][D][remote.raw:028]:   212, -213, 443, -209, 239, -213, 213, -239, 213, -213, 212, -451, 423, -227, 209, -440, 434, -231, 215, -215, 216, -247, 205, -236, 212, -439, 218, -215, 436, -227, 206, -236, 193, -461, 196, -239, 440, -433, 210, -216, 215, -215, 216, -240, 198, -230, 
[21:03:50][D][remote.raw:028]:   437, -438, 442, -230, 212, -427, 1520, -233, 212, -454, 417, -225, 229, -209, 237, -441, 210, -212, 213, -241, 420, -227, 214, -215, 219, -227, 209, -237, 193, -462, 418, -223, 221, -455, 436, -210, 214, -215, 216, -240, 225, -207, 236, -416, 218, 
[21:03:50][D][remote.raw:028]:   -215, 460, -200, 230, -235, 211, -441, 210, -213, 446, -424, 223, -215, 220, -227, 207, -237, 211, -216, 439, -419, 445, -228, 208, -438, 1535, -223, 226, -434, 435, -210, 215, -238, 216, -441, 206, -236, 212, -212, 442, -210, 239, -214, 214, -212, 
[21:03:50][D][remote.raw:028]:   239, -213, 212, -426, 445, -229, 208, -439, 440, -232, 214, -215, 216, -219, 228, -234, 211, -440, 194, -238, 437, -226, 206, -235, 212, -441, 210, -213, 445, -424, 223, -216, 220, -228, 207, -237, 211, -212, 442, -432, 440, -221, 218, -425, 1537, 
[21:03:50][D][remote.raw:028]:   -222, 218, -427, 435, -236, 214, -215, 216, -437, 231, -209, 237, -212, 441, -210, 213, -240, 213, -214, 212, -240, 213, -427, 444, -228, 207, -437, 435, -231, 215, -215, 216, -220, 229, -235, 211, -438, 210, -212, 446, -233, 212, -213, 213, -450, 199, 
[21:03:50][D][remote.raw:028]:   -222, 432, -462, 210, -213, 214, -240, 214, -214, 212, -214, 430, -443, 449, -206, 236, -415, 1534, -224, 228, -434, 435, -234, 190, -239, 216, -441, 207, -237, 193, -237, 438, -221, 218, -223, 207, -235, 212, -213, 239, -418, 440, -232, 212, -427, 
[21:03:50][D][remote.raw:028]:   444, -228, 207, -237, 212, -213, 239, -214, 213, -445, 207, -213, 428, -223, 216, -247, 206, -437, 236, -214, 418, -439, 232, -212, 213, -212, 231, -215, 218, -224, 434, -439, 440, -205, 214, -460, 1504, -231, 239, -423, 423, -230, 236, -212, 216, 
[21:03:50][D][remote.raw:028]:   -437, 222, -213, 219, -225, 434, -210, 216, -214, 217, -238, 218, -226, 207, -438, 433, -235, 212, -429, 443, -224, 228, -208, 236, -213, 212, -240, 214, -418, 234, -213, 429, -223, 216, -219, 229, -437, 210, -240, 418, -439, 233, -212, 213, -213, 238, 
[21:03:50][D][remote.raw:028]:   -204, 215, -218, 428, -436, 440, -231, 214, -435, 1528, -231, 212, -451, 421, -229, 209, -237, 212, -442, 210, -213, 240, -214, 420, -227, 214, -215, 220, -229, 235, -212, 216, -437, 440, -222, 199, -458, 436, -210, 214, -215, 240, -217, 225, -206, 
[21:03:50][D][remote.raw:028]:   237, -415, 231, -214, 435, -224, 227, -207, 237, -415, 236, -214, 445, -426, 235, -200, 216, -219, 227, -207, 236, -212, 416, -459, 424, -224, 215, -443, 1536, -220, 218, -424, 435, -236, 214, -215, 216, -437, 230, -209, 237, -192, 438, -226, 214, 
[21:03:50][D][remote.raw:028]:   -240, 198, -231, 235, -212, 213, -438, 441, -233, 212, -427, 444, -227, 206, -236, 212, -217, 214, -215, 215, -438, 231, -209, 439, -236, 212, -214, 239, -432, 195, -239, 440, -433, 210, -212, 241, -213, 214, -214, 213, -214, 447, -424, 443, -224, 227, 
[21:03:50][D][remote.raw:028]:   -439, 1511, -222, 221, -454, 438, -211, 212, -241, 213, -431, 232, -204, 214, -219, 428, -234, 212, -240, 213, -214, 213, -214, 240, -432, 418, -223, 222, -456, 435, -211, 212, -241, 213, -213, 213, -241, 213, -427, 222, -218, 427, -234, 212, -240, 
[21:03:50][D][remote.raw:028]:   214, -418, 234, -213, 429, -446, 221, -200, 231, -209, 238, -192, 238, -215, 436, -421, 458, -210, 216, -436, 1542, -210, 214, -460, 420, -229, 208, -238, 216, -437, 220, -214, 218, -225, 433, -210, 216, -215, 216, -238, 218, -228, 204, -438, 440, 
[21:03:50][D][remote.raw:028]:   -234, 212, -431, 443, -223, 226, -206, 237, -212, 216, -214, 217, -434, 225, -229, 435, -209, 217, -214, 240, -435, 228, -206, 437, -438, 210, -240, 214, -213, 214, -214, 213, -241, 427, -417, 451, -232, 212, -44 [... cut ...]
[21:03:51][D][remote.raw:028]:   -228, 207, -236, 416, -439, 446, -222, 219, -426, 1539, -222, 219, -426, 437, -235, 211, -239, 212, -431, 205, -230, 216, -219, 427, -234, 211, -238, 212, -213, 212, -239, 212, -430, 441, -223, 201, -459, 437, -236, 211, -212, 212, -239, 212, -212, 
[21:03:51][D][remote.raw:028]:   211, -451, 198, -249, 406, -234, 212, -240, 212, -445, 207, -212, 428, -443, 225, -228, 208, -236, 211, -217, 190, -240, 436, -447, 433, -208, 217, -437, 1542, -210, 215, -461, 420, -228, 208, -236, 193, -462, 197, -238, 218, -224, 433, -209, 217, 
[21:03:51][D][remote.raw:041]:   -215, 217, -239
```

dbuezas, 2023-07-26:
# Good news

* It looks like your wiring is indeed correct
* The bell sends the same signal a zillion times (that's good!)
* The signal looks complex but it isn't any special protocol

### You'll have to change the `idle` parameter of the remote_receiver to 1300us

(that's the absolute maximum would be 1500us or 1.5ms) 

### The raw code of your bell is:
`[-223, 226, -434, 435, -210, 215, -238, 216, -441, 206, -236, 212, -212, 442, -210, 239, -214, 214, -212, 239, -213, 212, -426, 445, -229, 208, -439, 440, -232, 214, -215, 216, -219, 228, -234, 211, -440, 194, -238, 437, -226, 206, -235, 212, -441, 210, -213, 445, -424, 223, -216, 220, -228, 207, -237, 211, -212, 442, -432, 440, -221, 218, -425]`

---

# How do I know

### Using [the tool](https://github.com/dbuezas/esphome-remote_receiver-oscilloscope), your original signal looks extremely long:
[image]

### But if you look closely, you'll see that it is a repeating pattern.

If I cut it each time there is a >1500us flank, I get this instead:

[image]

---

jwaes, 2023-07-27:
Thank you so much for your detailed explenation. I was indeed confused by this very long signal. 

It indeeds repeats a lot. But We can handle that inside esphome or in HA.

```
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:45][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state ON
[07:23:46][D][binary_sensor:036]: 'Doorbell': Sending state OFF
```

Thank you so much !

closed 2023-07-27 by jwaes

dbuezas, 2023-07-27:
G)ad to hear it worked!

The filter you need is this one:
https://esphome.io/components/binary_sensor/index.html#delayed-off