dbuezas/esphome-cc1101 #14: Help understanding and decoding signal for garage door opener

petrepa, 2023-08-12:
First, thank you for making this!

I am trying to replicate my garage door opener and for a single button press I get this output:

```
[12:45:49][D][remote.raw:028]: Received Raw: 453, -178, 213, -209, 229, -218, 219, -200, 228, -206, 234, -210, 236, -209, 211, -209, 237, -212, 207, -210, 236, -210, 236, -203, 218, -195, 249, -204, 233, -182, 245, -193, 242, -193, 241, -199, 245, -180, 1787, -395, 477, -384, 482, 
[12:45:49][D][remote.raw:028]:   -171, 486, -394, 478, -383, 282, -145, 265, -194, 240, -195, 251, -177, 259, -183, 268, -169, 265, -170, 265, -170, 476, -383, 267, -174, 260, -194, 466, -178, 258, -385, 283, -157, 486, -179, 262, -184, 254, -390, 477, -181, 261, -387, 488, -386, 472, 
[12:45:49][D][remote.raw:028]:   -173, 270, -176, 256, -182, 261, -387, 260, -185, 261, -157, 290, -157, 263, -184, 450, -199, 252, -402, 259, -182, 461, -181, 261, -402, 470, -399, 456, -408, 259, -183, 264, -159, 262, -184, 479, -173, 244, -200, 256, -384, 259, -192, 464, -173, 266, 
[12:45:49][D][remote.raw:028]:   -172, 252, -408, 485, -379, 268, -169, 489, -177, 257, -182, 244, -193, 242, -414, 470, -404, 257, -158, 487, -181, 262, -390, 471, -395, 251, -180, 487, -383, 284, -156, 264, -183, 454, -201, 240, -404, 479, -183, 261, -184, 236, -184, 262, -184, 262, 
[12:45:49][D][remote.raw:028]:   -375, 272, -172, 475, -181, 262, -184, 261, -389, 260, -181, 453, -394, 475, -206, 259, -387, 458, -411, 471, -396, 477, -384, 282, -169, 265, -169, 464, -402, 483, -383, 268, -170, 486, -399, 255, -183, 461, -193, 242, -193, 241, -194, 249, -406, 259, 
[12:45:49][D][remote.raw:028]:   -169, 265, -170, 266, -169, 469, -205, 235, -194, 240, -194, 241, -193, 243, -399, 483, -183, 246, -190, 243, -413, 250, -179, 485, -382, 268, -169, 487, -174, 250, -406, 258, -168, 265, -170, 266, -169, 269, -178, 258, -184, 261, -168, 267, -167, 462, 
[12:45:49][D][remote.raw:041]:   -199, 248, -179, 258, -385, 483, -401, 467, -199, 247, -405, 260, -167, 265, -169, 487, -374, 487, -401, 481, -172, 265, -391, 251, -179, 485, -383, 466, -204, 230, -195, 241, -200, 260, -4149, 227
[12:45:49][D][remote.raw:028]: Received Raw: 640, -182, 208, -207, 234, -209, 219, -217, 217, -219, 218, -220, 225, -206, 232, -209, 210, -210, 236, -210, 236, -183, 236, -210, 236, -183, 262, -183, 230, -194, 244, -200, 255, -182, 233, -195, 241, -194, 1767, -411, 453, -420, 474, 
[12:45:49][D][remote.raw:028]:   -174, 457, -409, 481, -398, 247, -170, 271, -178, 258, -181, 262, -183, 264, -156, 263, -182, 263, -183, 237, -183, 477, -396, 251, -180, 258, -183, 463, -205, 241, -389, 272, -171, 477, -181, 244, -192, 243, -414, 470, -178, 257, -409, 459, -414, 468, 
[12:45:49][D][remote.raw:028]:   -174, 253, -203, 233, -183, 244, -416, 246, -169, 265, -174, 278, -182, 259, -162, 465, -205, 236, -417, 248, -171, 468, -205, 235, -409, 459, -401, 470, -424, 252, -181, 264, -179, 236, -183, 486, -179, 261, -183, 236, -399, 248, -200, 455, -182, 244, 
[12:45:49][D][remote.raw:028]:   -196, 239, -412, 473, -407, 259, -169, 465, -200, 241, -194, 247, -178, 258, -386, 483, -402, 255, -178, 464, -201, 256, -384, 485, -389, 280, -157, 479, -394, 277, -154, 259, -182, 461, -207, 235, -416, 471, -173, 248, -198, 256, -182, 261, -168, 241, 
[12:45:49][D][remote.raw:028]:   -414, 246, -194, 473, -180, 260, -170, 241, -414, 248, -197, 469, -383, 487, -181, 240, -413, 471, -405, 459, -407, 475, -393, 249, -198, 255, -181, 460, -412, 455, -396, 272, -172, 477, -408, 260, -182, 468, -178, 261, -183, 229, -193, 244, -400, 256, 
[12:45:49][D][remote.raw:028]:   -182, 262, -184, 263, -156, 493, -179, 235, -211, 237, -177, 243, -195, 250, -408, 459, -209, 237, -183, 262, -402, 249, -169, 472, -405, 258, -183, 466, -206, 235, -402, 247, -195, 247, -182, 255, -183, 262, -158, 262, -184, 262, -183, 264, -157, 494, 
[12:45:49][D][remote.raw:041]:   -179, 229, -193, 273, -374, 482, -410, 462, -179, 262, -403, 244, -174, 278, -183, 458, -410, 465, -395, 472, -178, 259, -387, 285, -156, 485, -387, 471, -199, 248, -177, 257, -184, 260, -3968, 256
[12:45:50][D][remote.raw:028]: Received Raw: 640, -183, 210, -234, 208, -195, 217, -247, 190, -219, 241, -199, 229, -207, 234, -209, 236, -209, 211, -210, 236, -209, 237, -182, 236, -210, 236, -183, 229, -193, 245, -201, 256, -181, 262, -158, 263, -183, 1791, -374, 483, -411, 459, 
[12:45:50][D][remote.raw:028]:   -176, 461, -422, 456, -413, 253, -169, 265, -169, 267, -173, 277, -181, 262, -157, 269, -169, 265, -170, 265, -172, 475, -383, 283, -169, 265, -170, 462, -201, 256, -384, 260, -184, 461, -206, 237, -183, 263, -374, 496, -178, 257, -384, 485, -402, 467, 
[12:45:50][D][remote.raw:028]:   -172, 270, -177, 257, -182, 261, -391, 243, -193, 265, -171, 248, -179, 258, -183, 486, -182, 236, -405, 255, -179, 463, -201, 256, -384, 487, -386, 475, -393, 274, -177, 255, -182, 244, -168, 489, -175, 241, -200, 247, -407, 262, -165, 489, -175, 240, 
[12:45:50][D][remote.raw:028]:   -195, 249, -406, 459, -408, 266, -169, 487, -177, 256, -182, 260, -184, 238, -388, 481, -411, 249, -169, 495, -180, 234, -413, 457, -401, 272, -169, 467, -409, 259, -183, 263, -157, 493, -179, 263, -375, 469, -201, 255, -182, 260, -169, 241, -193, 242, 
[12:45:50][D][remote.raw:028]:   -415, 251, -179, 484, -183, 242, -193, 242, -388, 274, -178, 484, -381, 483, -175, 242, -411, 476, -408, 468, -396, 468, -399, 256, -183, 247, -190, 463, -395, 473, -407, 259, -183, 466, -412, 257, -183, 451, -199, 248, -177, 258, -182, 261, -385, 259, 
[12:45:50][D][remote.raw:028]:   -182, 265, -183, 261, -158, 477, -174, 270, -177, 257, -182, 261, -168, 241, -413, 468, -200, 228, -206, 234, -417, 243, -170, 486, -400, 256, -183, 462, -182, 244, -413, 271, -171, 249, -179, 259, -183, 262, -183, 267, -153, 264, -183, 264, -156, 481, 
[12:45:50][D][remote.raw:041]:   -173, 266, -172, 278, -383, 485, -386, 477, -174, 265, -399, 252, -182, 262, -184, 460, -412, 448, -423, 446, -206, 234, -412, 259, -183, 456, -395, 495, -177, 257, -181, 262, -183, 236, -3983, 218

```

Using the visualization tool we get this:
[image]

I've also tried to get a better understanding of what to do from this [issue](https://github.com/dbuezas/esphome-cc1101/issues/13), but I'm having a hard time wrapping my head around this. 

Do you see any immediate patterns and what the code could be for my garage door opener? My understanding is that the remote outputs several commands, the numbers after "Received Raw:" and that this is what gets plotted in the visualizer. I've tried to copy one of the lines and transmit this, but to no success.

dbuezas, 2023-08-12:
In a nutshell, the raw pattern specifies the length of each pulse and their spaces.
For example `600,-100,300`  means "there was a strong signal for 600us, followed by a 100us of silence and finally 300us of signal again.

In your case it looks like the transmitter sends the same signal 3 times. You should try pulsing it again to make sure the signals don't change each time you press the button.

To transmit, you can use the raw code directly (see EspHome remote_transmitter docs. There is a chance the door doesn't actually use this simple style of communication (pulses in a single frequency). For example it may be using multiple frequencies, in which case this won't work and you'll need to invest days of reverse engineering and reading.

When you copy and paste the pattern, make sure to add all the lines between two "Received Raw" (note how the EspHome output was many lines but the visualization only shows 3 signals)

dbuezas, 2023-08-12:
Oh, alternatively you could ask EspHome to try to find the protocol, using "all" instead of "raw"

petrepa, 2023-08-12:
Thank you for the explanation! 

As you pointed out, it seems like the transmitter is not outputting the same signal every time. There seems to be slight changes each time, which I guess implies a non-simple style of communication.

When changing to "all" instead of "raw" ESPHome finds the pronto protocol as follows:
```
[15:10:32][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 06C3
[15:10:35][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 06C3
[15:10:35][D][remote.pronto:238]: Received Pronto: data=0000 006D 009E 0000 0010 0007 0008 0009 0007 0009 0008 0009 0008 0008 0008 0009 0008 0009 0008 0009 0007 0009 0008 0008 0009 0008 0008 0009 0008 0008 0009 0007 000A 0008 0009 0008 0008 0008 0009 0008 0009 0008 0043 0011 0010 0011
[15:10:35][D][remote.pronto:240]: 0011 0008 0011 0010 0011 0010 0009 0008 0009 0008 0009 0007 0009 0007 0009 0009 0008 0008 0009 0008 0009 0008 0011 0010 0009 0008 0009 0008 0011 0009 0008 0011 0009 0008 0012 0007 0009 0008 0009 0010 0012 0008 0008 0011 0010 0011 0011 0009 0008 0008 0009 0008 0008 0011 0009 0008 0009 0007 0009 0008 0009 0008 0011 0009 0008 0008 0008 0009 0008 0010 0011 0008 0008 0010 0009 0008 0011 0010 000A 0007 0009 0008 0011 0009 0008 0008 0009 0008 0009 0008 0008 0010 0011 0010 0012 00
[15:10:36][D][remote.pronto:238]: Received Pronto: data=0000 006D 009F 0000 0007 0005 000B 0007 000A 0008 0008 0008 0009 0008 0009 0008 0009 0008 0009 0008 0009 0008 0009 0007 0009 0007 0009 0007 000A 0008 0009 0008 0009 0008 0008 0008 0009 0008 0008 0008 0009 0008 0008 0008 0044 0010
[15:10:36][D][remote.pronto:240]: 0011 0011 0011 0008 0011 0011 0011 0011 0009 0008 000A 0007 0009 0008 0009 0008 0009 0008 0009 0008 0008 0008 0009 0008 0011 0011 0009 0008 0009 0008 0011 0009 0008 0010 0009 0008 0012 0007 0009 0008 0009 0010 0012 0008 0008 0010 0011 0011 0011 0009 0008 0008 0009 0008 0008 0011 0009 0008 0008 0008 0009 0008 0009 0008 0011 0009 0008 0008 0009 0008 0009 0010 0012 0008 0009 0011 0009 0008 0011 0011 0008 0008 0009 0008 0011 0009 0008 0009 0009 0008 0008 0009 0008 0011 0011 00
[15:10:36][D][remote.pronto:238]: Received Pronto: data=0000 006D 009F 0000 0007 0004 000C 0006 0009 0007 0009 0009 0008 0008 0009 0008 0009 0008 0008 0008 0009 0008 0008 0008 0009 0008 0008 0008 0009 0008 0009 0008 0008 0009 0008 0008 0009 0008 0008 0008 0009 0008 0008 0008 0044 0010
[15:10:36][D][remote.pronto:240]: 0011 0011 0011 0007 0012 0010 0012 0010 0009 0009 0008 0008 0009 0007 000A 0008 0009 0008 0009 0008 0008 0008 0009 0008 0011 0011 0009 0008 0009 0008 0011 0009 0008 0011 0009 0008 0011 0008 0009 0008 0009 0010 0012 0008 0009 0011 0011 0010 0011 0009 0008 0008 0009 0009 0008 0011 0009 0008 0009 0007 0009 0007 000A 0007 0012 0009 0008 0008 0009 0008 0009 0011 0011 0009 0008 0011 0009 0008 0011 0011 0009 0008 0009 0008 0011 0009 0008 0008 0009 0008 0008 0009 0008 0010 0011 00
```

Double checked that I had done the right thing when copying the raw code for transmit. Sadly doesn't work, so guess it furthermore confirms a more advanced communication style. This is the output from ESPHome when the button is pressed:
```
[15:26:40][D][button:013]: 'Garage' Pressed.
[15:26:40][D][remote_transmitter:075]: Sending remote code...
[15:26:40][D][remote.raw:041]: Received Raw: 282
```
Don't know if the single number output from the received raw is as expected or what.

Guess I'll have to figure out something else to make my garage door smart, hehe

dbuezas, 2023-08-12:
Oh that's a pitty.
My guess is that it is either some multifrequency protocol or a so called "rolling-code" protocol (like a car fob). If you want to give it a last try, you can press the button multiple times and then copy all output to the visualiser.

petrepa, 2023-08-12:
After a bit more googling it seems like the my garage port is a Hormann BiSecur system, which uses AES encryption. Had a quick scroll through this video: https://www.youtube.com/watch?v=vPhaTZK2e8w&t=1439s

Think I'll put this Saturday project on hold for now... Thank you so much for your quick and kind help!

closed 2023-08-12 by petrepa

dbuezas, 2023-08-12:
Bummer. I hope you have other uses for the chip