dbuezas/esphome-cc1101 #16: ESP32-S2 Crashing on Transmit

lukas-holzner, 2023-12-18:
I have the module connected to a ESP32-S2. Recieving of signals works without any problems. Unfortuatly  when i press the button to send, Esphome crashes and reboots the esp. I don't detect any signals with my flipper zero either :(

```
sensor:
  - platform: custom
    lambda: |-
      auto 868 = new CC1101(
        36, // SCK
        37, // MISO
        35, // MOSI
        34, // CSN
        2, // GDO0
        200, // bandwidth_in_khz
        868.35 // freq_in_mhz
      );
      App.register_component(868);
      return {868};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: 2 # This is GDO0
    carrier_duty_percent: 100%

remote_receiver:
  - pin: 2 # This is GDO0
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump:
      - raw
binary_sensor:
  - platform: remote_receiver
    name: Garage
    raw:
      code: [-6278,417,-788,387,-788,383,-789,402,-391,790,-799,398,-389,788,-782,392,-788,387,-394,789,-390,791,-391,794,-793,387,-402,793,-386,788,-786,402,-386,788,-391,783,-786,379,-788,387,-390,795,-397,781,-786,387,-784,384,-788,387,-397,779,-399,785,-396,791,-785,388,-791,386,-780,383,-786,408,-784,385,-407,779,-388,794,-404,787,-784,387]
button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [-6278,417,-788,387,-788,383,-789,402,-391,790,-799,398,-389,788,-782,392,-788,387,-394,789,-390,791,-391,794,-793,387,-402,793,-386,788,-786,402,-386,788,-391,783,-786,379,-788,387,-390,795,-397,781,-786,387,-784,384,-788,387,-397,779,-399,785,-396,791,-785,388,-791,386,-780,383,-786,408,-784,385,-407,779,-388,794,-404,787,-784,387]
      - lambda: get_cc1101(transciver).endTransmission();
```

lukas-holzner, 2023-12-18:
It does receive signals.

dbuezas, 2023-12-19:
Unfortunately I haven't tried this myself on an esp32, less so in an s2. I don't see anything wrong in the yaml either. I suggest you look a the threads in this repo about esp32, and maybe also google the library this uses to communicate with the c1101

dbuezas, 2024-06-08:
Try the new cc1101 and yaml files. GDO2 not necessary in esp32 anymore.

closed 2024-06-08 by dbuezas