dbuezas/esphome-cc1101 #17: ESPHome 2023.12.1 can't use same pin for emit/receive

peroanjo, 2023-12-21:
Hello,

It seems like in ESPHome 2023.12.1 a new check happens and it is not possible to use the same pin for both the emitter and receiver, asi in the example cc1101.yaml:
[image]

What would be the best workaround for this?

Best regards

dbuezas, 2023-12-21:
https://esphome.io/guides/configuration-types#config-pin-schema

See allow_other_uses

peroanjo, 2023-12-21:
That's strange, after using the allow_other_uses option I keep receiving a strange raw message:

```
[19:12:25][I][remote.raw:041]:   -351, 250, -125, 75, -150, 100, -1026, 100, -1502, 700, -501, 726, -1277, 75, -75, 100, -1376, 101, -1477, 125, -125, 501, -550, 1101, -376, 75, -876, 75, -201, 476, -550, 51, -653, 97, -153, 173, -501, 75, -200, 75, -501, 300, -425, 501, -325, 551
[19:12:25][W][component:214]: Component remote_receiver took a long time for an operation (0.22 s).
[19:12:25][W][component:215]: Components should block for at most 20-30ms.
[19:12:25][I][remote.raw:028]: Received Raw: 75, -51, 75, -175, 100, -526, 75, -876, 51, -375, 325, -175, 701, -476, 125, -576, 325, -475, 51, -550, 151, -877, 400, -451, 75, -927, 275, -175, 476, -125, 125, -401, 250, -175, 651, -1052, 75, -300, 125, -826, 175, -751, 1302, -475, 100, 
[19:12:25][I][remote.raw:028]:   -225, 76, -575, 75, -150, 200, -651, 75, -75, 150, -201, 475, -877, 125, -75, 75, -125, 125, -76, 100, -2002, 400, -576, 250, -250, 175, -125, 175, -126, 400, -75, 150, -601, 125, -175, 225, -976, 125, -676, 300, -126, 275, -125, 75, -450, 226, -325, 
[19:12:25][I][remote.raw:028]:   651, -200, 225, -1677, 75, -151, 425, -1377, 75, -576, 75, -1102, 74, -126, 1001, -351, 75, -451, 100, -525, 1852, -125, 75, -326, 100, -150, 626, -150, 450, -175, 201, -150, 51, -225, 250, -100, 175, -250, 101, -200, 125, -125, 125, -200, 777, -100, 
[19:12:25][I][remote.raw:028]:   225, -175, 300, -651, 225, -100, 250, -225, 125, -1302, 100, -1027, 426, -100, 100, -100, 125, -551, 225, -326, 300, -575, 175, -651, 828, -425, 576, -475, 1577, -451, 200, -651, 100, -75, 575, -250, 276, -626, 100, -826, 726, -75, 276, -326, 100, 
[19:12:25][I][remote.raw:028]:   -751, 75, -125, 225, -175, 201, -1001, 225, -877, 75, -175, 400, -275, 275, -651, 175, -275, 75, -75, 1152, -826, 75, -225, 777, -325, 150, -251, 125, -300, 75, -275, 125, -75, 151, -625, 151, -1852, 326, -75, 575, -250, 75, -551, 175, -125, 525, -801, 
[19:12:26][I][remote.raw:028]:   225, -225, 100, -876, 150, -250, 325, -76, 926, -100, 100, -476, 200, -300, 175, -827, 125, -201, 475, -276, 100, -450, 125, -225, 476, -275, 125, -350, 501, -326, 575, -451, 126, -150, 175, -651, 100, -125, 101, -250, 125, -325, 150, -526, 100, -325, 
[19:12:26][I][remote.raw:028]:   175, -300, 150, -376, 75, -550, 125, -250, 75, -101, 150, -350, 100, -1577, 100, -125, 100, -401, 75, -775, 250, -451, 350, -176, 75, -1451, 250, -326, 100, -75, 75, -175, 801, -475, 75, -651, 200, -100, 75, -125, 51, -325, 726, -200, 75, -100, 75, 
[19:12:26][I][remote.raw:041]:   -275, 300, -251, 75, -601, 126, -1501, 125, -575, 201, -250, 100, -175, 250, -401, 125, -451, 226, -251, 100, -325, 176, -75, 100, -375, 151, -975, 200, -802, 275, -75, 150, -951, 1427
[19:12:26][W][component:214]: Component remote_receiver took a long time for an operation (0.19 s).
[19:12:26][W][component:215]: Components should block for at most 20-30ms.
[19:12:26][I][remote.raw:028]: Received Raw: 100, -150, 125, -225, 75, -251, 250, -100, 201, -100, 75, -150, 100, -550, 100, -175, 75, -51, 125, -150, 75, -175, 125, -300, 801, -551, 75, -701, 751, -1352, 150, -1402, 125, -626, 75, -75, 100, -200, 225, -977, 100, -501, 75, -1176, 125, 
[19:12:26][I][remote.raw:041]:   -426, 425, -175, 150, -100, 100, -200, 76, -325, 51, -625, 75, -400, 126, -100, 200, -250, 175, -325, 75, -526, 700, -200, 151, -1126, 1051, -75, 175, -100, 75, -326, 125, -700
[19:12:26][I][remote.raw:028]: Received Raw: -225, 200, -151, 75, -702, 75, -175, 800, -226, 75, -675, 100, -200, 727, -225, 375, -276, 201, -301, 75, -651, 725, -150, 75, -551, 75, -125, 75, -275, 276, -576, 175, -175, 75, -1001, 75, -100, 100, -226, 100, -150, 100, -400, 575, -176, 
[19:12:26][I][remote.raw:028]:   125, -175, 575, -175, 100, -301, 100, -125, 75, -401, 325, -726, 126, -550, 101, -100, 325, -275, 475, -176, 75, -425, 100, -350, 75, -576, 75, -100, 100, -225, 176, -275, 125, -76, 100, -125, 75, -376, 100, -501, 325, -376, 150, -175, 175, -75, 125, 
[19:12:26][I][remote.raw:028]:   -301, 100, -100, 75, -475, 476, -225, 75, -100, 126, -175, 451, -476, 200, -425, 100, -75, 851, -526, 75, -1051, 100, -426, 350, -100, 101, -350, 175, -576, 101, -775, 100, -225, 51, -300, 451, -125, 150, -400, 75, -251, 275, -100, 200, -151, 100, 
[19:12:26][I][remote.raw:041]:   -651, 300, -125, 75, -375, 125, -301, 125, -500
[19:12:26][W][component:214]: Component remote_receiver took a long time for an operation (0.08 s).
[19:12:26][W][component:215]: Components should block for at most 20-30ms.
[19:12:26][I][remote.raw:028]: Received Raw: -675, 326, -175, 526, -350, 76, -100, 325, -300, 75, -300, 175, -126, 100, -350, 126, -150, 201, -375, 75, -451, 175, -251, 75, -100, 1751, -201, 1602, -176, 625, -400, 276, -425, 100, -225, 225, -426, 75, -300, 200, -526, 125, -926, 100, 
[19:12:26][I][remote.raw:028]:   -275, 75, -426, 425, -100, 75, -225, 76, -200, 325, -801, 325, -225, 75, -101, 74, -276, 251, -75, 125, -75, 75, -100, 851, -175, 626, -475, 701, -250, 125, -575, 176, -652, 151, -150, 150, -375, 175, -752, 125, -451, 150, -350, 150, -100, 576, -100, 
```

I'm still able to receive my saved signal, any idea on what is happening here? This is my config yaml:

```
esphome:
  name: cc1101
  platform: ESP8266
  board: d1_mini
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  power_save_mode: HIGH

[... cut ...]
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  pin: # This is GDO0
    number: D1
    allow_other_uses: true
  carrier_duty_percent: 100%  

remote_receiver:
  pin: # This is GDO0
    number: D1
    allow_other_uses: true
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
  dump: raw
```

Thank you!

peroanjo, 2023-12-21:
nvm, it seems like the issue is only happening when I open the log connected durectly to my raspberry pi, so it must be something else.

dbuezas, 2023-12-21:
The `remote_receiver took a long time for an operation` is normal

closed 2024-06-08 by dbuezas