dbuezas/esphome-cc1101 #18: Custom Sensor Component to External Components.

erwan33fr, 2023-12-29:
Hello,

Would it be possible to move from "Custom Sensor Component" to "External Components" ?

https://esphome.io/components/sensor/custom.html

Custom components are deprecated, not recommended for new configurations and will be removed from ESPHome in a future release. 

Best regards,
Erwan

dbuezas, 2023-12-29:
Oh, good that they are deprecating it. I'll do that when I update my own board. Thanks for the heads up

gabest11, 2024-02-25:
Working on it! If anyone wants to test it, it's in my fork.

jmcruvellier, 2024-05-16:
> Working on it! If anyone wants to test it, it's in my fork.
> 
> https://github.com/gabest11/esphome-cc1101

Your code was working fine until I upgraded to ESPHome 2024.5.0 today.... now I'm getting : 

> [E][component:082]:   Component cc1101.sensor is marked FAILED

Any idea on what seems to be incompatible in your code with latest release of ESPHome?

gabest11, 2024-05-17:
I don't know, haven't upgraded yet. Can you test the latest version? I moved it to a top level component, as they suggested in the pull request to me. https://github.com/gabest11/esphome/tree/cc1101/esphome/components/cc1101

gabest11, 2024-05-17:
The pull request will probably never be approved, as the device cannot independently send and receive, and it needs that hack to switch direction.

dbuezas, 2024-06-08:
Try the new cc1101 and yaml files. GDO2 not necessary in esp32 anymore and it works on latest esphome

closed 2024-06-08 by dbuezas