dbuezas/esphome-cc1101 #21: ESPHOME ESP32 Issue

ginjac, 2024-01-10:
First off, thanks for this code dbuezas! I think it is really great! 
I have this working ok on an esp8266, but have had some issues with esp32 devices. If I try to flash with the default cc1101.h the esp32 bootloops. If I add the following to cc1101.h its stops bootlooping and boots fine // noInterrupts();  // interrupts(); Receive works perfect transmit does not. Checked with RTL-SDR no transmission. Tried adding the GDO2 pin and putting receive on that to seperate transmit and receive, same result. Saw this info in another post. Any advice would be appreciated. See below...

  void beginTransmission() {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.SetTx();
    pinMode(_GDO0, OUTPUT);
    // noInterrupts();
  }
  void endTransmission() {
    // interrupts();
    pinMode(_GDO0, INPUT);
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.SetRx();
    ELECHOUSE_cc1101.SetRx();  // yes, twice

dbuezas, 2024-01-11:
I haven't tried it on an esp32 myself yet so I'm not sure what the problem may be. The implementation of the receiver and transmitter are completely different between both platforms, so there may well be some assumption I made for the 8266 that doesn't work on the esp32.
I hope someone else with a working esp32 can help you

dbuezas, 2024-06-08:
Try the new cc1101 and yaml files. GDO2 not necessary in esp32 anymore.

closed 2024-06-08 by dbuezas