dbuezas/esphome-cc1101 #22: Can't run it on ESPHome 2023.12.6

KrzysztofHajdamowicz, 2024-01-16:
Hi!

It was a while since I used my 433 sniffer so I decided to start from fresh firmware.
Unfortunately, looks like device can't boot right now, logs from esphome-flasher
[flashing.log](https://github.com/dbuezas/esphome-cc1101/files/13950701/flashing.log)

I'm attaching my YAML configuration:
```yaml
# https://github.com/dbuezas/esphome-cc1101
esphome:
  name: 433-receiver
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"
esp8266:
  board: nodemcuv2

logger:
  level: DEBUG
api:
  encryption:
    key: "[...]"
ota:
  password: "[...]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "433-Receiver Fallback Hotspot"
    password: "[...]"

sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        D5, // SCK
        D6, // MISO
        D7, // MOSI
        D3, // CSN
        D1, // GDO0
        200, // bandwidth_in_khz
        433.92 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};

    sensors:
      id: transciver_1
      name: "RSSI"
      unit_of_measurement: dBm
      entity_category: diagnostic

remote_receiver:
  - id: receiver_1
    pin:
      number: D1 # This is GDO0
      # allow_other_uses: True
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump: all
```

Photos of device:
[image]
[image]

dbuezas, 2024-01-16:
Try removing everything (particularly the logger) except the cc1101 code to make sure you aren't getting out of RAM.
Then maybe try removing the cc1101 code to confirm the esp board is in working order

dbuezas, 2024-06-08:
I'm running it w/o issues

closed 2024-06-08 by dbuezas