dbuezas/esphome-cc1101 #24: ESP32 needs separate GDO2 and cannot disable interrupts

gabest11, 2024-02-23:
I was trying to get the bottom of this and basically the code from #2, and when noInterrupts(); / interrupts(); is called then it will crash with this stack, so you may want to remove those with ESP32.

```
0x40091d5b: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c line 182
0x40090a7f: vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/cpu_ll.h line 39
0x4008fcb6: xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 1688
0x400fd78b: rmt_write_items at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/rmt.c line 1165
0x400e142e: esphome::remote_transmitter::RemoteTransmitterComponent::send_internal(unsigned int, unsigned int) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_vector.h line 805
0x40198f9d: esphome::remote_base::RemoteTransmitterBase::send_(unsigned int, unsigned int) at src/esphome/components/remote_base/remote_base.cpp line 160
0x400eca83: esphome::remote_base::RemoteTransmitterActionBase ::play(ArduinoJson6185_D1::ObjectConstRef) at src/esphome/components/remote_base/remote_base.h line 118

```

dbuezas, 2024-02-24:
Solved by your PR, thanks!

closed 2024-02-24 by dbuezas

tichris0, 2024-06-06:
I'm trying to make my implementation work.  I'm using a WROVER-ESP-32 and am running into some issues.  If I run the code, it crashes (see last comment in #28 https://github.com/dbuezas/esphome-cc1101/issues/28#issuecomment-2146809152).  Reading this thread, it looks like this crash is caused by the interrupt code (which I agree with since disabling that makes the crash go away, however I go from code that used to send something and crashed to code that now sends nothing according to my SDR radio but doesn't crash.  Any insight into what's going on here?  Here's my file:

```
// https://github.com/dbuezas/esphome-cc1101

#ifndef CC1101TRANSCIVER_H
#define CC1101TRANSCIVER_H

#include <ELECHOUSE_CC1101_SRC_DRV.h>

int CC1101_module_count = 0;
#define get_cc1101(id) (*((CC1101 *)id))

class CC1101 : public PollingComponent, public Sensor {
  int _SCK;
  int _MISO;
  int _MOSI;
  int _CSN;
  int _GDO0; // TX (and also RX if ESP8266)
  float _bandwidth;
  
  float _moduleNumber;
  int _last_rssi = 0;

  void setup() {
    ELECHOUSE_cc1101.setSpiPin(_SCK, _MISO, _MOSI, _CSN);

    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setGDO0(_GDO0);
    ELECHOUSE_cc1101.setCCMode(0);
    ELECHOUSE_cc1101.setModulation(2);
    ELECHOUSE_cc1101.setMHZ(_freq);
    ELECHOUSE_cc1101.setDeviation(47.60);
    ELECHOUSE_cc1101.setRxBW(_bandwidth);
    ELECHOUSE_cc1101.setDRate(3);
    ELECHOUSE_cc1101.setPA(0);
    ELECHOUSE_cc1101.setPktFormat(3);
    ELECHOUSE_cc1101.setDcFilterOff(1);
    
    ELECHOUSE_cc1101.SetRx();
    pinMode(_GDO0, INPUT);
  }

 public:
  float _freq;
  CC1101(int SCK, int MISO, int MOSI, int CSN, int GDO0, 
         float bandwidth, float freq)
      : PollingComponent(100) {
    _SCK = SCK;
    _MISO = MISO;
    _MOSI = MOSI;
    _CSN = CSN;
    _GDO0 = GDO0;
    _bandwidth = bandwidth;
    _freq = freq;
    _moduleNumber = CC1101_module_count++;
  }

  void beginTransmission() {
    ELECHOUSE_cc1101.SetTx();
    pinMode(_GDO0, OUTPUT);
    delayMicroseconds(100);
//    noInterrupts();
  }
  void endTransmission() {
//    interrupts();
    ELECHOUSE_cc1101.SetTx();
    ELECHOUSE_cc1101.SetRx();
    pinMode(_GDO0, INPUT);

  }
  void setBW(float bandwidth) {
    ELECHOUSE_cc1101.setRxBW(bandwidth);
  }
  void setFreq(float freq) {
    ELECHOUSE_cc1101.setMHZ(freq);
  }
  bool rssi_on;
  void update() override {
    int rssi = 0;
    if (rssi_on) {
      rssi = ELECHOUSE_cc1101.getRssi();
    }
    if (rssi != _last_rssi) {
      publish_state(rssi);
      _last_rssi = rssi;
    }
  }
};

#endif
```

gabest11, 2024-06-06:
Since you have esp32, try to use two pins (send on gdo0, receive on gdo2) and remove the direction flipping (pinMode).

tichris0, 2024-06-06:
> Since you have esp32, try to use two pins (send on gdo0, receive on gdo2) and remove the direction flipping (pinMode). That's actually in the title of this discussion.

That did not work for me.  It sent nothing at all when I tried doing things with the stock code that's checked in.  It only started working more so when I starting flipping on GDO0.

dbuezas, 2024-06-06:
I think the cc1101 needs to be told to use gdo0 to receive and gdo2 to transmit, and that's missing in the PR that added esp32 support. I suggest looking in the docs of the SmartRC-CC1101-Driver-Lib docs