dbuezas/esphome-cc1101 #28: Receives but doesn't send commands

tichris0, 2024-06-01:
I was able to configure my WROVER-B device, have it register to Home Assistant and I have no problem receiving data according to the serial console, but for some reason it doesn't actually send anything according to my SDR & Universal Radio Hacker.  Given that I can receive, I'm guessing that the issue is with my configuration.  My GDO pins re slightly different given that I'm using a slightly different esp32.  My understanding is that GDO2 isn't used in this case as the CC1101 chip can only Tx or Rx at any point in time but not both.

My own C app written in Arduino & using the same library works, so I know that the hardware is connected fine and can work as expected.

When I use the remote control I see the following which closely resembles the code I want to send (aka: with some errors):
```
[0;32m[I][remote.raw:041]: Received Raw: 413, -653, 388, -662, 387, -659, 365, -677, 367, -319, 733, -650, 385, -657, 384, -659, 380, -316, 745, -642, 401, -295, 729, -681, 382[0m
```

When I press the button in Home Assistant:
```
[0;36m[D][button:010]: 'Garage' Pressed.[0m
[0;36m[D][sensor:094]: 'transciver': Sending state -78.00000  with 0 decimals of accuracy[0m
```

My yaml configuration:
```
# https://github.com/dbuezas/esphome-cc1101
esphome:
  name: cc1101
  platform: ESP32
  board: nodemcu-32s
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"

wifi:
  ssid: <hidden>
  password: <hidden>
  fast_connect: true
  power_save_mode: HIGH

logger:
api:
ota:

sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        18, // SCK
        19, // MISO
        23, // MOSI
        5, // CSN
        2, // GDO0
        4, // GDO2
        48, // bandwidth_in_khz
        303.886 // freq_in_mhz
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - pin: GPIO2 # This is GDO0
    carrier_duty_percent: 100%

remote_receiver:
  - pin: GPIO4 # This is GDO2
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump:
      - raw
binary_sensor:
  - platform: remote_receiver
    name: Garage
    raw:
      code: [346,-692,346,-692,346,-692,346,-692,346,-346,692,-692,346,-692,346,-692,346,-692,346,-692,346,-346,692,-692,346,-1038]
button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [346,-692,346,-692,346,-692,346,-692,346,-346,692,-692,346,-692,346,-692,346,-692,346,-692,346,-346,692,-692,346,-1038]
      - lambda: get_cc1101(transciver).endTransmission();

```

dbuezas, 2024-06-01:
Looking at the h file, I think `ELECHOUSE_cc1101.setGDO(GDO0, GDO2);` is missing for esp32. I never tested that contribution by a user so there may be a bug there. See https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master

tichris0, 2024-06-02:
Hmm I'm not familiar enough with esphome's yaml and how that the whole thing works.  Is that something that should be added to the yaml?  In theory I can compare my function calls to my C code to know what's missing, but not really familiar with esphome at a Code level.

dbuezas, 2024-06-02:
The cc1101.h file.

tichris0, 2024-06-04:
Yah I concur with the discussion follow up that you had #25 .  I managed to make mine work by changing your code to essentially ignore the USE_ESP32 specific code.  Moreover, I've removed all of the ESP32 conditional code and treated it using the regular path and that managed to work.  I can send & receive no problem now.  Caveat however is that whenever I send something.

Although I can send & receive, sending a stream always results in an immediate crash

```
[23:57:38]Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1).
[23:57:38]
[23:57:38]Core  1 register dump:
[23:57:38]PC      : 0x4008ec14  PS      : 0x00060d35  A0      : 0x8008d932  A1      : 0x3ffb2340
WARNING Decoded 0x4008ec14: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c:166 (discriminator 1)
[23:57:38]A2      : 0x3ffb2e44  A3      : 0x3ffb88f0  A4      : 0x00000004  A5      : 0x00060d23
[23:57:38]A6      : 0x00060d23  A7      : 0x00000001  A8      : 0x3ffb88f0  A9      : 0x00000018
[23:57:38]A10     : 0x3ffb88f0  A11     : 0x00000018  A12     : 0x3ffc4394  A13     : 0x00060d23
[23:57:38]A14     : 0x007bf118  A15     : 0x003fffff  SAR     : 0x00000020  EXCCAUSE: 0x00000006
[23:57:38]EXCVADDR: 0x00000000  LBEG    : 0x400899b0  LEND    : 0x400899bb  LCOUNT  : 0x00000000
[23:57:38]
[23:57:38]
[23:57:38]Backtrace:0x4008ec11:0x3ffb23400x4008d92f:0x3ffb2360 0x4008ccb2:0x3ffb2380 0x400e70ea:0x3ffb23c0 0x400da67e:0x3ffb2400 0x4015d06d:0x3ffb2450 0x400e065d:0x3ffb2470 0x4015d49b:0x3ffb2490 0x4015d4af:0x3ffb24b0 0x4015d4cd:0x3ffb24d0 0x400d7725:0x3ffb24f0 0x400d30b3:0x3ffb2510 0x4015cb61:0x3ffb2530 0x400d6a35:0x3ffb2550 0x400d3625:0x3ffb25e0 0x400d71c2:0x3ffb2680 0x4015d269:0x3ffb2790 0x4015d315:0x3ffb27b0 0x400dd634:0x3ffb27d0 0x400dfae2:0x3ffb2800 0x400e507d:0x3ffb2820
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008ec11: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c:166 (discriminator 1)
WARNING Decoded 0x4008d92f: vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3566
WARNING Decoded 0x4008ccb2: xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:1688
WARNING Decoded 0x400e70ea: rmt_write_items at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/rmt.c:1165
WARNING Decoded 0x400da67e: esphome::remote_transmitter::RemoteTransmitterComponent::send_internal(unsigned int, unsigned int) at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/components/remote_transmitter/remote_transmitter_esp32.cpp:121
WARNING Decoded 0x4015d06d: esphome::remote_base::RemoteTransmitterBase::send_(unsigned int, unsigned int) at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/components/remote_base/remote_base.cpp:162
WARNING Decoded 0x400e065d: esphome::remote_base::RemoteTransmitterBase::TransmitCall::perform() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome\components\remote_base/remote_base.h:118
 (inlined by) esphome::remote_base::RemoteTransmitterActionBase<>::play() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome\components\remote_base/remote_base.h:257
WARNING Decoded 0x4015d49b: esphome::Action<>::play_complex() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/automation.h:125
WARNING Decoded 0x4015d4af: esphome::Action<>::play_next_() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/automation.h:155
 (inlined by) esphome::Action<>::play_complex() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/automation.h:126
WARNING Decoded 0x4015d4cd: esphome::ActionList<>::play() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/automation.h:203
 (inlined by) esphome::Automation<>::trigger() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/automation.h:241
 (inlined by) esphome::Trigger<>::trigger() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/automation.h:98
 (inlined by) esphome::button::ButtonPressTrigger::ButtonPressTrigger(esphome::button::Button*)::{lambda()#1}::operator()() const at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome\components\button/automation.h:23
 (inlined by) std::_Function_handler<void (), esphome::button::ButtonPressTrigger::ButtonPressTrigger(esphome::button::Button*)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at c:\users\chris\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
WARNING Decoded 0x400d7725: std::function<void ()>::operator()() const at c:\users\chris\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
 (inlined by) esphome::CallbackManager<void ()>::call() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/core/helpers.h:482
 (inlined by) esphome::button::Button::press() at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/components/button/button.cpp:12
WARNING Decoded 0x400d30b3: esphome::api::APIConnection::button_command(esphome::api::ButtonCommandRequest const&) at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/components/api/api_connection.cpp:833
WARNING Decoded 0x4015cb61: esphome::api::APIServerConnection::on_button_command_request(esphome::api::ButtonCommandRequest const&) at A:\Downloads\cc1101\.esphome\build\cc1101/src/esphome/components/api/api_pb2_service.cpp:1221
[... cut ...]
[23:57:40]Backtrace:0x4008ed90:0x3ffbea6c |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008ed90: compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:288
```

dbuezas, 2024-06-07:
I just tested an esp32 with the code as is and both sending and receiving work fine for me.

dbuezas, 2024-06-08:
I found a way to use a single pin in the esp32, I'll update the repository soon

dbuezas, 2024-06-08:
Try this, you only need to connect GDO0 in an esp32 and let me know.
If it works well for you I'll check compatibility with esp8266 and maybe add a variant.
```cpp
// cc1101.h
// https://github.com/dbuezas/esphome-cc1101

#ifndef CC1101TRANSCIVER_H
#define CC1101TRANSCIVER_H

#include <ELECHOUSE_CC1101_SRC_DRV.h>

#include "esphome/components/remote_transmitter/remote_transmitter.h"

int CC1101_module_count = 0;
#define get_cc1101(id) (*((CC1101*)id))

class CC1101 : public PollingComponent, public Sensor {
  int _SCK;
  int _MISO;
  int _MOSI;
  int _CSN;
  int _GDO0;  // TX and also RX
  float _bandwidth;
  esphome::remote_transmitter::RemoteTransmitterComponent* _remote_transmitter;
  float _moduleNumber;
  int _last_rssi = 0;

  void setup() {
    pinMode(_GDO0, INPUT);
    ELECHOUSE_cc1101.addSpiPin(_SCK, _MISO, _MOSI, _CSN, _moduleNumber);
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setRxBW(_bandwidth);
    ELECHOUSE_cc1101.setMHZ(_freq);
    ELECHOUSE_cc1101.SetRx();
  }

 public:
  float _freq;
  CC1101(int SCK, int MISO, int MOSI, int CSN, int GDO0, float bandwidth, float freq,
         esphome::remote_transmitter::RemoteTransmitterComponent* remote_transmitter)
      : PollingComponent(100) {
    _SCK = SCK;
    _MISO = MISO;
    _MOSI = MOSI;
    _CSN = CSN;
    _GDO0 = GDO0;
    _bandwidth = bandwidth;
    _freq = freq;
    _moduleNumber = CC1101_module_count++;
    _remote_transmitter = remote_transmitter;
  }

  void beginTransmission() {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.SetTx();
    pinMode(_GDO0, OUTPUT);
    _remote_transmitter->setup();
  }
  void endTransmission() {
    digitalWrite(_GDO0, 0);
    pinMode(_GDO0, INPUT);
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.SetRx();
  }
  void setBW(float bandwidth) {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.setRxBW(bandwidth);
  }
  void setFreq(float freq) {
    ELECHOUSE_cc1101.setModul(_moduleNumber);
    ELECHOUSE_cc1101.setMHZ(freq);
  }
  bool rssi_on = false;
  void update() override {
    int rssi = 0;
    if (rssi_on) {
      ELECHOUSE_cc1101.setModul(_moduleNumber);
      rssi = ELECHOUSE_cc1101.getRssi();
      if (rssi != _last_rssi) {
        publish_state(rssi);
        _last_rssi = rssi;
      }
    }
  }
};

#endif
```

```yaml
# cc1101-esp32.yaml
esphome:
  name: fan-test
  friendly_name: Fan test
  includes:
    - cc1101.h
  libraries:
    - SPI
    - "SmartRC-CC1101-Driver-Lib"

esp32:
  board: esp32dev
  
wifi:
  ssid: !secret wifi_name
  password: !secret wifi_pass
  fast_connect: true
  power_save_mode: HIGH

logger:
  level: VERBOSE
api:
ota:

web_server:
  port: 80

number:
  - platform: template
    max_value: 812
    min_value: 58
    step: 1
    mode: slider
    optimistic: true
    unit_of_measurement: "kHz"
    name: BW
    on_value:
      then:
        - lambda: get_cc1101(transciver).setBW(x);
  - platform: template
    # if your cc1101 board is tuned for ~320MHz
    # min_value: 300 
    # max_value: 348

    # if your cc1101 board is tuned for ~433MHz
    min_value: 378
    max_value: 464

    # if your cc1101 board is tuned for ~868Hz
    # min_value: 799
    # max_value: 928
    
    step: .001
    mode: box
    optimistic: true
    unit_of_measurement: "MHz"
    name: FREQ
    on_value:
      then:
        - lambda: get_cc1101(transciver).setFreq(x);

sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new CC1101(
        18, // SCK
        19, // MISO
        23, // MOSI
        5, // CSN
        32, // GDO0
        200, // bandwidth_in_khz
        433.92, // freq_in_mhz
        id(transmitter) // id of remote_transmitter
      );
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      id: transciver
      internal: true
remote_transmitter:
  - id: transmitter
    pin: 
      number: GPIO32 # This is GDO0
      allow_other_uses: true
    carrier_duty_percent: 100%

remote_receiver:
  - id: receiver
    pin: 
        number: GPIO32 # This is GDO0
        allow_other_uses: true
      # on the esp8266 use any of D1,D2,D5,D6,D7,Rx
      # Don't use D3,D4,D8,TX, boot often fails.
      # Can't be D0 or GPIO17 b/c no interrupts
    dump:
      - raw
binary_sensor:
  - platform: remote_receiver
    name: Garage
    raw:
      code: [990,-330,330,-990,990,-330,330,-990,330,-990,330,-990,330,-990,330,-990,330,-990,990,-330,330,-990,990,-330,330,-990,330,-990,330,-990,330,-990,330,-990,330,-990,990,-330,990,-330,330,-990,330,-990,990,-330,330,-990,330]
switch:
  - platform: template
    name: "RSSI"
    entity_category: diagnostic
    lambda: return get_cc1101(transciver).rssi_on;
    turn_on_action:
      lambda: get_cc1101(transciver).rssi_on = true;
    turn_off_action:
      lambda: get_cc1101(transciver).rssi_on = false;

button:
  - platform: template
    name: Garage
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          code: [990,-330,330,-990,990,-330,330,-990,330,-990,330,-990,330,-990,330,-990,330,-990,990,-330,330,-990,990,-330,330,-990,330,-990,330,-990,330,-990,330,-990,330,-990,990,-330,990,-330,330,-990,330,-990,990,-330,330,-990,330]
          repeat:
            times: 3
            wait_time: 4.733ms
      - lambda: get_cc1101(transciver).endTransmission();
```

dbuezas, 2024-06-08:
Try the new cc1101 and yaml files. GDO2 not necessary in esp32 anymore.

closed 2024-06-08 by dbuezas

tichris0, 2024-06-09:
Thanks.  I'll give that a try.