dbuezas/esphome-cc1101 #29: Signal with 65 bits

TheChatty, 2024-06-09:
I have a remote which sends a signal with 65 bits - the protocol definition is like:
```
RCSwitchBase(5200,  600, 600, 200, 200, 600, false)
```

The original receiver accepts this signal even when I send only the first 64 bits. Thus I can define a button using this protocol and send button presses.

I can define a binary_sensor using rc_switch_raw. This way I am notified about presses from the original remote.

But due to [this problem](https://github.com/esphome/issues/issues/5538) I'd like to receive these signals with a defined protocol. At the moment rc_switch can't see those signals.

dbuezas, 2024-06-09:
Maybe you can use an "on_raw" automation, you get the received signal in a variable
https://esphome.io/components/remote_receiver.html

TheChatty, 2024-06-09:
The device being controlled is a projection canvas. Having it pulling down takes 50s. It can be stopped at any point.

How to build a yaml which has a variable `canvas_extension_height` in percent? I should react on original remote signals and esphome signals. There should be another boolean variable `canvas_moving'`.

dbuezas, 2024-06-09:
You could use a time based cover template: https://esphome.io/components/cover/time_based

closed 2024-06-12 by dbuezas