dbuezas/esphome-cc1101 #32: need help with ceiling fan codes

magnetus26, 2025-01-05:
hello David et al.
1st thanks for this project, its a great project.
I have a cealing fan with an RF unit that im trying to bring to home assistant.
the controller P/N is 2bhlf-tyx0001 and it operates at 433.92Mhz
hxxps://fccid.io/2BHLF-TYX0001
I built my trasnmitter but need help understanding my logs and RF trasnmit code.

need help on knowing what im looking at.
file ...logs (1).txt is when i depress a button on the fan trasmitter
file ...logs.txt is when I depress Garabe button on the web server page

please see attached for logs.

[logs_cc1101-example_logs (1).txt](https://github.com/user-attachments/files/18310436/logs_cc1101-example_logs.1.txt)
[logs_cc1101-example_logs.txt](https://github.com/user-attachments/files/18310437/logs_cc1101-example_logs.txt)

dbuezas, 2025-01-05:
Pase those here to visualize them https://dbuezas.github.io/esphome-remote_receiver-oscilloscope/

magnetus26, 2025-01-05:
> Pase those here to visualize them https://dbuezas.github.io/esphome-remote_receiver-oscilloscope/

done. but Idont know what im looking at.
[image]

that is my light ON/OFF button on the fan.
how can I paste this into the esphome code to control it?
i know that the frequency is 433.92Mhz , how can I determine the bandwith and the  remote_transmitter.transmit_raw: code?
thanks
Miguel

dbuezas, 2025-01-05:
Hi, you are seeing the raw signal patterns. The fact that they look like exact repetitions is promising. I suggest you read the esphome section about remote receiver and emitter, there's nothing I can add to that

magnetus26, 2025-01-05:
will do that
thank you
Miguel

closed 2025-01-05 by magnetus26