dbuezas/esphome-cc1101 #34: custom platform no longer works with esphome 2025.2.x and up

mikje, 2025-03-03:
The yaml wont compile in esphome 2025.2 and up because of a breaking change with the custom platform

mcmasterp, 2025-03-16:
yeah same boat. I tried to research how to switch it to external component but I just couldn't grasp it.

billmaterial, 2025-03-16:
I got it working with this version https://github.com/gabest11/esphome-cc1101
I needed to modify the code as I have very specific requirements to work with my kinetic switches but if you just want general use then this should work.

awdark, 2025-03-24:
Thank you the gabest11 repository worked for me.

In case it helps anyone, with the current version of ESPHome I needed to use the "allow_other_uses" for the pin and sensor values. Don't mind my weird GPIO choices used for the ethernet version.  The sample yaml there also does not return anything back to HA so this version returns the decimal version of the binary rc_switch output.  Transmit via raw code in HA  actions not seem to work.  

```
spi:
  clk_pin: GPIO14
  miso_pin: GPIO4
  mosi_pin: GPIO15

remote_transmitter:
  - pin: 
      number: GPIO2 # GDO0
      allow_other_uses: true
    carrier_duty_percent: 100%
    

remote_receiver:
  pin: 
    number: GPIO35 # GDO2
    allow_other_uses: true
  dump:
    - rc_switch
    - raw

  tolerance: 50%
  #buffer_size: 2kb
  filter: 250us
  idle: 4ms
  on_rc_switch: 
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            code: !lambda 'return x.code;'

sensor:
  - platform: cc1101
    id: transceiver
    cs_pin: GPIO12
    gdo0: 
      number: GPIO2
      allow_other_uses: true
    gdo2: 
        number: GPIO35
        allow_other_uses: true
    bandwidth: 200
    frequency: 433920
    rssi:
      name: "wt32 cc1101 rssi"
    lqi:
      name: "wt32 cc1101 lqi"

```